bigtreetech/SKR-2 #1: How much is Neopixel Led support in SKR 2?

Foxcor, 2021-04-05:
Previous boards used the optional BIGTREETECH DCDC5V V1.0 module for 30 LEDs. How much support on the new board?

Portzal, 2021-04-09:
I would also be interested in how they have done away with having the 5v module as a option.

E-Studios, 2021-04-26:
So far i heard it was 30 just like the SKR 1.4 Addon module but they integrated it in the SKR 2 board.

thisiskeithb, 2021-04-26:
> So far i heard it was 30 just like the SKR 1.4 Addon module but they integrated it in the SKR 2 board.

There's now an integrated AOZ1284PI in the motherboard ([schematic](https://github.com/bigtreetech/SKR-2/blob/master/Hardware/BIGTREETECH%20SKR%202-SCH.pdf)), so no addon module is needed.

[image]

Seminole81, 2021-08-03:
Is this due to the Amps going through the board? For instance, I have 144 Neopixel on a 1 meter strip, works out to just over 8A as they are 5V. Can you run all 144 through the SKR board, or do you need an external supply with a common ground to handle more than the 30?