bigtreetech/SKR-2 #2: Sensor less homing jumper documentation.

Portzal, 2021-04-05:
I have looked through the manual and pin .pdf but see no definite information on where to jump the diag pin.some more details would be much appreciated.

thisiskeithb, 2021-04-05:
They are located under the drivers. I modified the [top view drawing](https://github.com/bigtreetech/SKR-2/blob/master/Hardware/BIGTREETECH%20SKR%202-top.pdf) and circled them in blue below:

[image]

Portzal, 2021-04-05:
Thanks for the response, I was thinking they would be the likely pins but could not find anything to back up my assumption.

thisiskeithb, 2021-04-05:
I just updated my reply with the photo from the manual so it's clearer.

> I was thinking they would be the likely pins but could not find anything to back up my assumption.

BigTreeTech added a lot of features to this form factor, so they were limited on where to place jumpers.

ph1lloops, 2021-04-21:
Am I correct with this: putting the jumper on the board on that place = I want to use sensorless homing

thisiskeithb, 2021-04-21:
> Am I correct with this: putting the jumper on the board on that place = I want to use sensorless homing

Yes.

closed 2021-04-28 by Portzal