bigtreetech/SKR-2 #4: Not working

calderon67, 2021-04-16:
I have installed the given marlin with the appropriate configuration changes for my 3d printer but it is not working. It has a really strange behavior. It apparently moves and works normal. But suddenly, while bed leveling with hot bed (70ºC) some how X motor stopped, but the rest keep on working well. It is a prusa with TMC2130 on SPI mode. I have tried changing motor, changing wire, and changing driver, but I had no luck.

I asked for M122 command before and after failure, and it reports strange data.
This is working normal
[image]

This is after failure, without sending any command but M122 S0
[image]

When it gets into failure, the X motor just stop moving.

As SKR 2 has not been included in marlin official repository I was not able to test another marlin version just to check.

Can anybody help me? I feel frustrated and I do not know what else to do...

thanks a lot

thisiskeithb, 2021-04-16:
It will be supported upstream soon: https://github.com/MarlinFirmware/Marlin/pull/21503

calderon67, 2021-04-16:
Does it means that today it can not be used?

calderon67, 2021-04-16:
and also, soon means matter of days or maybe matter of months?