bigtreetech/SKR-2 #5: pop up message  "All LOW"

vMeph, 2021-04-18:
on the touchscreen on startup pop ups a message saying "All LOW"
is this related with the board?
on other board never had this error

radek8, 2021-04-24:
Check fuses for 12/24V

vMeph, 2021-04-26:
not sure but i think this might have to do with the pin out of the board?
```
Driver registers:
		X	0xC0:0C:00:00
		Y	0xC0:0C:00:00
		Z	0xC0:0C:00:00
		Z2	0x00:00:00:00	 Bad response!
		E	0xC0:0C:00:00

Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... Error: All LOW
Testing E connection... OK
```
any toughts?

Rodov, 2021-04-26:
It looks like Z2 didn't connected

vMeph, 2021-04-26:
i have switch driver and error still on Z2 so im assuming is hardware pins or something

vMeph, 2021-04-26:
in sk1.4 for exemple have on software serial
```
  #define Z2_SERIAL_TX_PIN                 P1_01
  #define Z2_SERIAL_RX_PIN                 P1_01
```
in skr2 pin out witch have the same layout that pinout  for Z2 wont exist

vMeph, 2021-04-26:
ups i think i have discovery the problem im using on drive for the 2 steppers and have defined 2 Z drives grrrr
thanks, closing this for now

closed 2021-04-26 by vMeph