bigtreetech/SKR-2 #6: Stuck at marlin-bootscreen

Github-Anni, 2021-04-20:
Use the provided marlin version from BTT github repository and compiled errorfree in visual studio.
But when connecting (usb)power the led screen only shows the marlin bootscreen and stuck.
Any idee whats going wrong

[image]

[configuration-files.zip](https://github.com/bigtreetech/SKR-2/files/6346333/configuration-files.zip)

radek8, 2021-04-24:
Try this:
comment
// # define CR10_STOCKDISPLAY
uncomment
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
Use two ribbon cables and connect EXP1 and EXP2.
Do not connect EXP3 on the display.

Github-Anni, 2021-04-26:
unfortunately same issue even more worse with  #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
[image]

Github-Anni, 2021-04-26:
changed from TMC5160 Steppers to TMC2208v3, had them laying around ;)
now board is booting as expected but not with the TMC5160 Steppers
changed the following settings in conf.h
#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208
#define E0_DRIVER_TYPE TMC2208
to
#define X_DRIVER_TYPE  TMC5160
#define Y_DRIVER_TYPE  TMC5160
#define Z_DRIVER_TYPE  TMC5160
#define E0_DRIVER_TYPE TMC5160

and in conf_adv.h i changed the following:
disable SPI 
  //#define TMC_USE_SW_SPI
  //#define TMC_SW_MOSI       -1
  //#define TMC_SW_MISO       -1
  //#define TMC_SW_SCK        -1

whats wrong with the TMC5160?

Github-Anni, 2021-05-06:
ok could fix the issue with the new marlin 2.0.8 release