bigtreetech/SKR-2 #7: Make this repository to be a fork of oficial Marlin

xorza, 2021-04-21:
Current problem is that I cannot migrate my printer to SKR 2 board because I cannot get diff from this repo to Marlin master and apply this diff onto my current Marlin version.

Guys this repository has only one commit, and I cannot find Marlin head commit for this repo. this is so inconvenient and unprofessional. If you want to sell your boards you should make it easy to get a software for them, I dont have a week of my life to spare trying to fix this for you

thisiskeithb, 2021-04-21:
BTT’s board repos contain more than a copy Marlin, so that’s why they aren’t setup that way. You can find BTT’s fork of Marlin here: https://github.com/bigtreetech/Marlin

Even if this repo was setup to just host Marlin, the fork would get stale quickly, so you’d want to use the main Marlin repo anyway. This board and driver protection feature will be merged upstream soon: https://github.com/MarlinFirmware/Marlin/pull/21503

xorza, 2021-04-21:
I re-read my previous message and my tone feels harsh, I apologize. Now I understand that you didnt want this repo to only have Marlin in it.

Thanks for the link to the Marlin PR, this is very helpful, somehow I missed it. I will take changes from that branch and apply onto my Marlin.

Closing the issue

closed 2021-04-21 by xorza