bigtreetech/SKR-2 #10: "E0-Axis anti pluged" ERROR

rabby7, 2021-04-27:
I compiled the firmware as supposed to. everything is plugged well. Using an Ender-3. 
Now I get "E0-Axis anti pluged" ERROR.
Found nothing on the internet 
How should I proceed?

rabby7, 2021-04-28:
I am thinking of disabeling the test for anti reverse in the firmware, but afraid it might do harm… Is there anybody with this board that work?

rabby7, 2021-04-29:
https://www.youtube.com/watch?v=WyThpaba_bk

rabby7, 2021-04-29:
SENDING:M122
		X	Y	Z	E
Address		0	0	0	0
Enabled		false	false	false	false
Set current	800	800	800	800
RMS current	1436	1436	1436	1436
MAX current	2025	2025	2025	2025
Run current	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31
CS actual	0/31	0/31	0/31	0/31
PWM scale
vsense		0=.325	0=.325	0=.325	0=.325
stealthChop	false	false	false	false
msteps		256	256	256	256
interp		false	false	false	false
tstep		0	0	0	0
PWM thresh.
[mm/s]
OT prewarn	false	false	false	false
pwm scale sum	0	0	0	0
pwm scale auto	0	0	0	0
pwm offset auto	0	0	0	0
pwm grad auto	0	0	0	0
off time	0	0	0	0
blank time	16	16	16	16
hysteresis
 -end		-3	-3	-3	-3
 -start		1	1	1	1
Stallguard thrs	0	0	0	0
uStep count	0	0	0	0
DRVSTATUS	X	Y	Z	E
sg_result	0	0	0	0
stst		*	*	*	*
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0x00:00:00:00	 Bad response!
		Z	0x00:00:00:00	 Bad response!
		E	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW
//action:notification TMC CONNECTION ERROR

mjg2688, 2021-04-29:
Are you using TMC 2209 steppers?  If so there is a known issue that BTT is addressing, for some reason the board of toasting 2209 drivers.  They reached out to me on a Reddit post, awaiting a response now.  Per their advice, discontinue use of this board until the fix is released.  Weather the fix is a software change or them sending out new boards I do not know at this time.

rabby7, 2021-04-29:
Thanks

בתאריך יום ה׳, 29 באפר׳ 2021, 20:42, מאת mjg2688 ***@***.***

mjg2688, 2021-04-29:
> Thanks
>
> בתאריך יום ה׳, 29 באפר׳ 2021, 20:42, מאת mjg2688 ***@***.***

I misspoke, it was comments on a video this morning on YouTube.  Took some screenshots.  

[image]
[image]

rabby7, 2021-04-30:
The solution they sent me is working well

rabby7, 2021-04-30:
[SKR 2+TMC2209.pdf](https://github.com/bigtreetech/SKR-2/files/6403176/SKR.2%2BTMC2209.pdf)

closed 2021-04-30 by rabby7