bigtreetech/SKR-2 #11: LCD doesn't work

schvvalbe, 2021-04-28:
Use the provided marlin version from BTT github repository and compiled errorfree in visual studio.
But when connecting (usb)power showing this (picture). With other wotherboard this lcd works correctly
Any idee whats going wrong
[image]
[Configuration.zip](https://github.com/bigtreetech/SKR-2/files/6390657/Configuration.zip)

b-desconocido, 2021-05-06:
Under
`#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`
I added these defines
`#define ST7920_DELAY_1 DELAY_NS(0) 
#define ST7920_DELAY_2 DELAY_NS(900) 
#define ST7920_DELAY_3 DELAY_NS(0) `
I have no clues what these delays are and how exactly they works, but increasing the second one up to 900 solved my issues

Github-Anni, 2021-05-06:
had same issue.
switched to Marlin 2.0.8 and fixed it

brotherchris, 2021-07-01:
I'm having the same type of issue. The LCD works, but nothing on EXP2 works. Not sure what's going on. I have tired 2.0.8, the BTT version and 2.0.9.1, all the same. Same code, my SKR 1.3 works fine. Seems like it has to be a bad pin assignment.

Github-Anni, 2021-07-01:
[BTT TFT to BTT Mainboards 2021-May.pdf](https://github.com/bigtreetech/SKR-2/files/6749793/BTT.TFT.to.BTT.Mainboards.2021-May.pdf)
Try the correct setting as explained in the attached PDF

brotherchris, 2021-07-01:
I am using those settings. I have multiple SKR2 boards. It's very strange. The screen works fine on the SKE REVA board, but doesn't in the REVB boards.