bigtreetech/SKR-2 #12: #error "M100 - unsupported CPU"

deathvenom-369, 2021-04-28:
cant seem to compile. This is the source pulled directly from git-hub unmodified.

> Executing task: C:\Users\***\.platformio\penv\Scripts\platformio.exe run --environment BIGTREE_SKR_2 <

Processing BIGTREE_SKR_2 (platform: ststm32@~12.0; board: genericSTM32F407VGT6; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F407VGT6.html
PLATFORM: ST STM32 (12.0.0) > STM32F407VG (192k RAM. 1024k Flash)
HARDWARE: STM32F407VGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (stlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 0.0.0
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TMCStepper> 0.7.1
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.5.0
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
|-- <EEPROM> 2.0.1
Building in release mode
Compiling .pio\build\BIGTREE_SKR_2\src\src\gcode\calibrate\M100.cpp.o
Marlin\src\gcode\calibrate\M100.cpp:112:4: error: #error "M100 - unsupported CPU"
  112 |   #error "M100 - unsupported CPU"
      |    ^~~~~
Marlin\src\gcode\calibrate\M100.cpp: In function 'int check_for_free_memory_corruption(const char*)':
Marlin\src\gcode\calibrate\M100.cpp:202:29: error: 'free_memory_start' was not declared in this scope
  202 |   char *start_free_memory = free_memory_start, *end_free_memory = free_memory_end;
      |                             ^~~~~~~~~~~~~~~~~
Marlin\src\gcode\calibrate\M100.cpp:203:11: error: 'end_free_memory' was not declared in this scope; did you mean 'dump_free_memory'?
  203 |   int n = end_free_memory - start_free_memory;
      |           ^~~~~~~~~~~~~~~
      |           dump_free_memory
Marlin\src\gcode\calibrate\M100.cpp: In static member function 'static void GcodeSuite::M100()':
Marlin\src\gcode\calibrate\M100.cpp:342:8: error: 'free_memory_end' was not declared in this scope
  342 |   if (!free_memory_end) free_memory_end = sp - MEMORY_END_CORRECTION;
      |        ^~~~~~~~~~~~~~~
Marlin\src\gcode\calibrate\M100.cpp:342:48: error: 'MEMORY_END_CORRECTION' was not declared in this scope
  342 |   if (!free_memory_end) free_memory_end = sp - MEMORY_END_CORRECTION;
      |                                                ^~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
Archiving .pio\build\BIGTREE_SKR_2\lib0c2\libLiquidCrystal.a
*** [.pio\build\BIGTREE_SKR_2\src\src\gcode\calibrate\M100.cpp.o] Error 1
============================================================================================================================================================================== [FAILED] Took 16.29 seconds ==============================================================================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
BIGTREE_SKR_2  FAILED    00:00:16.285
========================================================================================================================================================================= 1 failed, 0 succeeded in 00:00:16.285 ========================================================================================================================================================================= 
The terminal process "C:\Users\****\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'BIGTREE_SKR_2'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

Rincewind08, 2022-05-12:
same

> Executing task in folder Firmware: C:\Users\User\.platformio\penv\Scripts\platformio.exe run --environment BIGTREE_GTR_V1_0 <

Processing BIGTREE_GTR_V1_0 (board: marlin_BigTree_GTR_v1; platform: ststm32@~12.1; framework: arduino)
--------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_BigTree_GTR_v1.html
PLATFORM: ST STM32 (12.1.1) > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash    
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200819 (1.9.0)      
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)      
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing TMCStepper @ ~0.7.3

Unpacking  [------------------------------------]    0%
Unpacking  [------------------------------------]    1% 
Unpacking  [#-----------------------------------]    3%
Unpacking  [#-----------------------------------]    4% 
Unpacking  [##----------------------------------]    6% 
Unpacking  [##----------------------------------]    8% 
Unpacking  [###---------------------------------]    9%
Unpacking  [####--------------------------------]   11% 
Unpacking  [####--------------------------------]   13%
Unpacking  [#####-------------------------------]   14% 
Unpacking  [#####-------------------------------]   16% 
Unpacking  [######------------------------------]   18% 
Unpacking  [#######-----------------------------]   19% 
Unpacking  [#######-----------------------------]   21% 
Unpacking  [########----------------------------]   22%
Unpacking  [########----------------------------]   24%
Unpacking  [#########---------------------------]   26%
Unpacking  [##########--------------------------]   27% 
Unpacking  [##########--------------------------]   29%
Unpacking  [###########-------------------------]   31% 
Unpacking  [###########-------------------------]   32%
Unpacking  [############------------------------]   34% 
Unpacking  [############------------------------]   36% 
Unpacking  [#############-----------------------]   37% 
Unpacking  [##############----------------------]   39% 
Unpacking  [##############----------------------]   40% 
Unpacking  [###############---------------------]   42% 
Unpacking  [###############---------------------]   44% 
Unpacking  [################--------------------]   45% 
Unpacking  [#################-------------------]   47%
Unpacking  [#################-------------------]   49% 
Unpacking  [##################------------------]   50%
Unpacking  [##################------------------]   52%
Unpacking  [###################-----------------]   54% 
Unpacking  [####################----------------]   55% 
Unpacking  [####################----------------]   57% 
Unpacking  [#####################---------------]   59% 
Unpacking  [#####################---------------]   60%
Unpacking  [######################--------------]   62% 
Unpacking  [#######################-------------]   63%
Unpacking  [#######################-------------]   65% 
Unpacking  [########################------------]   67%
Unpacking  [########################------------]   68%
Unpacking  [#########################-----------]   70% 
Unpacking  [#########################-----------]   72%
Unpacking  [##########################----------]   73%
Unpacking  [###########################---------]   75% 
Unpacking  [###########################---------]   77% 
Unpacking  [############################--------]   78% 
[... 523 lines cut ...]
Marlin\src\gcode\calibrate\M100.cpp:202:29: error: 'free_memory_start' was not declared in this scope
  202 |   char *start_free_memory = free_memory_start, *end_free_memory = free_memory_end;
      |                             ^~~~~~~~~~~~~~~~~   
Marlin\src\gcode\calibrate\M100.cpp:203:11: error: 'end_free_memory' was not declared in this scope; did you mean 'dump_free_memory'?
  203 |   int n = end_free_memory - start_free_memory;  
      |           ^~~~~~~~~~~~~~~
      |           dump_free_memory
Marlin\src\gcode\calibrate\M100.cpp: In static member function 'static void GcodeSuite::M100()':
Marlin\src\gcode\calibrate\M100.cpp:342:8: error: 'free_memory_end' was not declared in this scope
  342 |   if (!free_memory_end) free_memory_end = sp - MEMORY_END_CORRECTION;
      |        ^~~~~~~~~~~~~~~
Marlin\src\gcode\calibrate\M100.cpp:342:48: error: 'MEMORY_END_CORRECTION' was not declared in this scope       
  342 |   if (!free_memory_end) free_memory_end = sp - MEMORY_END_CORRECTION;
      |                                                ^~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\BIGTREE_GTR_V1_0\src\src\gcode\calibrate\M100.cpp.o] Error 1
============= [FAILED] Took 48.46 seconds =============

Environment       Status    Duration
----------------  --------  ------------
BIGTREE_GTR_V1_0  FAILED    00:00:48.463
======== 1 failed, 0 succeeded in 00:00:48.463 ======== 
The terminal process "C:\Users\User\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'BIGTREE_GTR_V1_0'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

hcet14, 2022-05-14:
Which µC? F407 or F429?