bigtreetech/SKR-2 #13: Connecting RPi

Bubba2017, 2021-04-29:
SKR2 Documentation indicates connecting Raspberry Pie for the SKR 1.4
The two boards appear to be completely different.

No row of pins at the edge of the board?

thisiskeithb, 2021-04-29:
The diagram applies to the SKR 2 as well. The row of pins you see on the left side of the diagram is a Raspberry Pi, not an SKR board:

[image]

Bubba2017, 2021-04-29:
Could you show the connections to the SKR2,  If you connecting to the TFT connector the labels don't match.
+5v, GND, PAS, PAIO, RST
Which is connected to TXD0 on the RPi and which is RXD0 on RPi?

aabeuge, 2021-05-03:
Refer to the schematic in the hardware folder.  On SKR-2 PA9 is TX and PA10 is RX.  So pin 10 (RXD0) of RPi would go to the middle pin (PA9) of tft connector on SKR-2 and pin 8 of RPI would go to the 4th pin from left (PA10) on tft connector.

Bubba2017, 2021-05-04:
That is the information that was missing.  Thanks.  
BTT please add the infromation for aabeuge to the documentation.

printur33, 2021-12-10:
Hey @Bubba2017 how did this work out for you and what RPi version are you running?

cjmalchow, 2021-12-12:
Question, are there other ports on the SKR2 that can be used for the TX and RX (PA9 and PA10)? Those ports are in use for my configuration as they are needed for the display. Thanks!

PixelatedPolyurethane, 2022-09-23:
> That is the information that was missing. Thanks. BTT please add the infromation for aabeuge to the documentation.

A bit late but i just did one. enjoy!

[image]