bigtreetech/SKR-2 #14: Overheating drivers?

Portzal, 2021-04-30:
Is it true there is a issue with TMC2209's drivers overheating on this board?

rabby7, 2021-04-30:
[SKR 2+TMC2209.pdf](https://github.com/bigtreetech/SKR-2/files/6403184/SKR.2%2BTMC2209.pdf)
Check this file

mjg2688, 2021-04-30:
> I guess there is one connection per tmc2209 from MGND to GND necessary?
> 
> Does this also happen with tmc2226 drivers?
> 
> 
> Overall it looks like bigtreetech doesn't test their new boards with one of the most popular stepper driver (tmc2209). Very disappointing. 

Not the issue is only with the 2209.  And even with R&D things can be missed especially when things go from prototype to production.  For all we know is is a defect caused from manufacturing, not due to BTT lack of testing.  And really their reaching out to their customers the correct and replace boards, what more can you ask for?  

Early adoption of new products/technologies has ALWAYS come with potential drawbacks.  If you don't like it maybe wait a few months next time instead of rushing in to pick up the newest thing without understanding the potential drawbacks.

rabby7, 2021-04-30:
I think one wire is enough for all
I used 2 wires just to be safe for all 4 stepper drivers.
Having trouble now woth the TFT35 and octoprint but I believe it is not related to this problem.
I get "No media" statement. I print through Prontaface - working well.

Portzal, 2021-04-30:
Reading through the documents BTT has released on the issue it looks like it refers to all TMC drivers, not just 2209's.

oechslein, 2021-04-30:
There is now an official response from bigtreetech: https://docs.google.com/document/d/1IeKgfE2WIDjqH1fx5Yg7n1FOHVwhDFmDlZ-7QMlOEV0/edit?usp=sharing

And it looks like all drivers with sensorless homing are affected.