bigtreetech/SKR-2 #16: Uncertain which rev board I have

aabeuge, 2021-05-06:
My board has one green QC Pass 02 sticker on it. Based on the BTT Bulletin I should have a rev. b board but their example rev. B photo shows a grey sticker.

What FET should rev. B. boards have?  I'm kinda hesitant to power this up with TMC2209's until more certain my board isn't affected.

EsserPrototyping, 2021-05-06:
Only one sticker = Rev A

aabeuge, 2021-05-06:
Thanks, I thought it seemed counterintuitive for the example rev b. board to have two stickers, "QC Pass 09" and "QC Pass 02" and state to people that if we have QC Pass 02 that we have rev. B board.  So they managed to botch that as well lol.

I'll bet this will confuse btt staff into denying requests for replacement boards because they think they have rev. B boards.

EsserPrototyping, 2021-05-07:
Oh yes, that´s pretty confusing! Mine has only one green sticker with QC Pass 09.
In the end.. if your mosfet is "G090N06" it´s Rev A. Don´t know why they need to make it more complicated.

closed 2021-06-27 by aabeuge

arcenik, 2022-02-21:
Hi,

I've just bought a SKR2 board and the revision is not marked.
Is the one sticker rule still valid in 2022 ?
The mofset is a HY1904, is it rev.A rev.B or something else ?

[image]

Thanks

JaHoSto, 2022-04-08:
Ich habe die gleiche Konstellation. 
Ein grüner Aufkleber mit 08 und ebenfalls den oben genannten HY1904 

Rev A oder B  ?

EsserPrototyping, 2022-04-08:
G090N06 = bad
HY1904 = good
That's all you need to know.

binhamed84, 2022-04-09:
what is your processor Serial number?

STM32F407 = Rev A

STM32F429 = Rev B

JaHoSto, 2022-04-09:
HY1904 + STM32F429

Thanks..  ;-)