bigtreetech/SKR-2 #17: Please add warning and information about Rev A to github and inform customers.

EsserPrototyping, 2021-05-06:
I think it´s very important that you add a warning and the corresponding links to your github SKR-2 mainfolder / description.

Also I noticed, that - till now - I never received any information about the problem via email - and I ordered via your own webstore.

If I did not inform myself via youtube or other media, I would still think my board is totally fine and I wonder how many drivers are still getting killed by other customers who never heard about the problem.

EsserPrototyping, 2021-05-07:
In case someone don´t want to google to get the information..

Announcement: https://docs.google.com/document/d/1IeKgfE2WIDjqH1fx5Yg7n1FOHVwhDFmDlZ-7QMlOEV0/edit?fbclid=IwAR2C08blcj-n2OIn2bAAzp0HBsOXjyxAex5Mft9HWQ30yx3eu4dJ4VlRkX8

Check revision:
https://docs.google.com/document/d/1swmc4HvP9vxrxV2b9LVGa_I7GLTQGogQns7CfMYCckA/edit

Since there seems to be some Rev A boards with "QC Pass 2" Sticker in green, but no second sticker... check the mosfet - if it´s "G090N06" it is REV A!

thisiskeithb, 2021-05-12:
When compiling with the latest upstream Marlin bugfix, you will now prompted with a warning & a link to BigTreeTech's announcement.

MasT3r-A, 2021-05-21:
> In case someone don´t want to google to get the information..
>
> Announcement: [...]

My mosfet is a HY1904 - does it mean i have a rev. B board? 

On the backside of the board i have two labels (one green QC PASS 08 & one silver QC PASS 01).

I am a bit confused because in the documentation it says "sticker with QC PASS 02".

mjg2688, 2021-05-21:
> My mosfet is a HY1904 - does it mean i have a rev. B board?
>
>
>
> On the backside of the board i have two labels (one green QC PASS 08 & one silver QC PASS 01).
>
>
>
> I am a bit confused because in the [...]
This seems self explanatory to me but to put it simply YES you have a rev B board which is the FIXED version.

MasT3r-A, 2021-05-21:
> This seems self explanatory to me but to put it simply YES you have a rev B board which is the FIXED version.

Can you explain why it is self explanatory to you .. ? because of the MOSFET? i didn't find any info about the rev B MOSFET ..

mjg2688, 2021-05-21:
> > > > "check the mosfet - if it´s "G090N06" it is REV A!"
> 
> > > 
> Can you explain why it is self explanatory to you .. ? because of the MOSFET? i didn't find any info about the rev B MOSFET .. 
> 
> 
> 
> 
Safe bet if you DO NOT have the "G090N06" MOSFET on your board then it is safe to use, not sure how much more clear this can be?  😔

MasT3r-A, 2021-05-21:
The official paper which was provided by bigtreetech does not say anything about the wrong/correct type of MOSFET. Only one guy, in this case <EsserPrototyping> is writing something about the MOSFET... i mean i am thankful that he posted the answer to my problem but where is evidence that this is correct?! any sources, papers, anything?!?!?

MasT3r-A, 2021-05-21:
Ok .. i found the MOSFET HY1904C2 in the schematic.. sorry for triggering you but i wanted to be sure and didn't want to fry my TMC2130s.

mjg2688, 2021-05-21:
> The official paper which was provided by bigtreetech does not say anything about the wrong/correct type of MOSFET. Only one guy, in this case <EsserPrototyping> is writing something about the [...]

Yep, here is a screenshot of the schematic off the SKR 2 GitHub repo 
[image]

pyro2927, 2021-06-06:
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/pins/stm32f4/pins_BTT_SKR_V2_0_REV_A.h#L26-L27 seems to explicitly prevent you from compiling until you read and clear the `#error`. Might be good to merge that in.

gordo3di, 2022-01-11:
Just an FYI. We've had 7 boards now which all had the old G090N06 MOSFETS (labeled QC pass 2) fail. Our order of 50 boards had various QC pass labeled with the foil sticker, green sticker etc... It looks like a bunch of stuff was just mislabeled and randomly mixed in despite the assurance it was all rev B. 

Basically during heat up the MOSFET would remain on until max temp and it would remain this way indefinitely. For us it was always E0 but this is most likely because we sell more single extruders.

We finally got the new SKR v2 with the F429 ARM chip and I can confirm they have  HY1904 A1K00LA1N (F429) vs old G090N06 A1A03LS1N (F407) and also the G060P04 MOSFET (F429) vs old G045P03 (F407). Now I am about to compile marlin 2.0.9.3 which includes the new chip. Just posting this here in case it helps someone down the road. Biqu (or bigtreetech or whatever they are called these days) has been extremely unhelpful during this entire problem and it has cost us a lot of money as well as pissed off customers. Basically they just ignored all our emails.

jaysuk, 2022-01-11:
@gordo3di why not com over to flymaker boards and join us on discord and use RRF. I would be quite happy to arrange a couple of boards for you to try https://discord.gg/uS97Qs7