bigtreetech/SKR-2 #19: heatbreak fan where to connect and program in Marlin?

Simsela, 2021-05-18:
Hello,
Can't get any further with my SKR 2 board, this wasn't a problem with my SKR 1.4!

Will the fan on the SKR2 board also be connected to the HE1 and how can I get it controlled in Marlin? or do I have to connect it to a FAN port and how can I control it depending on the temperature?

can someone please help me?

Thanks in advance!

aabeuge, 2021-05-18:
All of the SKR-2 pins are defined (aliased) in /Marlin/src/pins/stm32f4/pins_BTT_SKR_V2_0.h which will show you the alias/pin name to look for in the configuration files (look at line numbers 209 through 214).  With that you can then reference line number 458 in Configuration_adv.h which appears to be the sec tion that corresponds to your question.  I'm not very familiar yet with PWM controlled fans in Marlin as I'm just another user but hopefully this points you in the right direction.

Simsela, 2021-05-18:
> 
> 
> I'm not sure if HE1 can be used for a fan like ramps does but it looks like Fan0 or Fan1 is probably the one to use if you want control of it in marlin. All of the SKR-2 pins are defined (aliased) in /Marlin/src/pins/stm32f4/pins_BTT_SKR_V2_0.h which will show you the alias/pin name to look for in the configuration files (look at line numbers 209 through 214). With that you can then reference line numbers 458 through 560) in Configuration_adv.h which appears to be the section that corresponds to your questions. Fan0 is probably the part cooling fan by default so maybe Fan1 might be better for the heatbreak fan. I'm not very familiar yet with PWM controlled fans in Marlin as I'm just another user but hopefully this points you in the right direction.

Hello,
Now I understand it  and it worked! Many Thanks!

Mito73a, 2021-06-21:
Hi, can you can explain to me? 
The fan for the material always goes to 100% and I cannot control it. Where did you connetted it? on fan0? and in marlin where did you comment?
Thank's!!!!!

LTSpenny, 2021-07-22:
Could you share how you set it up to work?

StArL0rd84, 2021-07-23:
> Hello,
> Now I understand it and it worked! Many Thanks!

Please help us understand exactly how you made it work.
In Configuration_adv.h I've set E0_AUTO_FAN_PIN to PB6 (Fan1) and COOLER_AUTO_FAN_PIN to PB7 (Fan0).
Preheated printer, But still, the pins refuse to activate. 
If i switch the pins around (PB6 Fan0) (PB7 Fan1) VsCode throws error trying to build.
Cannot further calibrate the printer untill this is reolved.

hoax78, 2021-07-27:
Found this, hope it's help a bit!
https://3daddict.com/marlin-custom-fan-pins/

matthgyver, 2021-08-10:
I've same problem. After lot of tests, there is my solution:

Into ``Marlin/src/pins/stm32f4/pins_BTT_SKR_V2_0.h`` file :
* Comment this two lines:
```
// #define CONTROLLER_FAN_PIN                  PB5   //FAN1 motor CONTROLLER_FAN
// #define CONTROLLER_FAN_PIN2                 PB6   //FAN2 motor CONTROLLER_FAN
```
* Uncomment this two lines:
```
#define FAN1_PIN                            PB6   // Fan1
#define FAN2_PIN                            PB5   // Fan2
```

Into ``Marlin/Configuration_adv.h`` file:
* Comment this line:
```
// #define USE_CONTROLLER_FAN
```

* Set the auto fan pin:
```
#define E0_AUTO_FAN_PIN FAN1_PIN
```