bigtreetech/SKR-2 #23: Please update 3D step file to include the additional sd card Usb ports

marcobox, 2021-05-26:
Hello,

I am in the process of designing a case for the SKR 2.0 to use with my Ender 3v2.

I was very happy when I found the 3D cad file in the repository. However upon inspecting the content I notices that the STEP file is missing the SD card slot and the additional USB A connector to the left. Would it be possible to add them? They are very valuable to design the cutout in the case to access them.

jon-harper, 2021-06-01:
At a glance, the fuses are also missing from the STEP model. I mostly care about the connectors as well, though.

DominikPalo, 2022-01-01:
I'm also designing a new enclosure for SKR 2 and having the same issue with missing parts (USB A/SD) required to model cutouts properly. So could you please update the STEP file? It would really help...

[image]