bigtreetech/SKR-2 #25: DIAG Pin for sensorless homing

jejdouay, 2021-05-28:
I have found on the documentation how to use the jumper to allow the sensorless homing.

I am using klipper and need to know what are the pin name for the diag pin for X and Y ( and other if there is ?

Can you help me out ?

aabeuge, 2021-05-28:
The pins file in marlin is helpful to look at for any pins.  Check out: Firmware/Marlin-bugfix-2.0.x/Marlin/src/pins/stm32f4/pins_BTT_SKR_V2_0.h
Lines 54 and 55.  PC1 and PC2

Exeu, 2021-07-04:
`#define X_DIAG_PIN                          PC1   // X-STOP
#define Y_DIAG_PIN                          PC3   // Y-STOP
#define Z_DIAG_PIN                          PC0   // Z-STOP
#define E0_DIAG_PIN                         PC2   // E0DET
#define E1_DIAG_PIN                         PA0   // E1DET`

therainkin, 2022-01-07:
thanks @Exeu I came here to look for that as well.  Strange it's not in the SKR2 documentation anywhere.