bigtreetech/SKR-2 #28: SKR V2 Jumpers

MarkV3082, 2021-06-17:
I would like to know what the configuration for the onboard jumpers under the Driver chips is. Specifically what pin is linked to what. I have seen various recommendations for jumper setings for the FYSTEC v 3.0 driver but  no explaation of what these pins actually ARE.  thanks in advance for any assistance

ps-bond, 2021-06-17:
You've seen the silkscreen on the underside I presume?

MarkV3082, 2021-06-17:
I didn't really pay too much attention to it before installing the board. I just assumed that the jumper pin settings would be as per the ones I had seen on the net. I have the printout of the bottom of the board from Bigtreetech but it doent explain what the all various pins are connected too.  Are you saying its all there on the bottom but not on the Bigtree docs and I have to remove the board to copy it down?. Surely its listed somewhere??

ps-bond, 2021-06-17:
No idea - mine's sat on the bench beside me.

Top (main power connector on LHS) to bottom:

MOSI
SCK
CS
MISO

Obviously those are SPI designations - this doesn't tell you what they're overloaded as (e.g. UART mode).  The connector is on the schematic.  I think those are your only choices though?

closed 2021-07-15 by MarkV3082