bigtreetech/SKR-2 #35: Help! SKR 2 Error: Failed to initialise WiFi module: bad reply format version

Gramm79, 2021-07-02:
Good day. after the firmware I wanted to activate wi-fi. Throws this error. Which is from BTT, which is from teamgloomy.
Can anyone understand what the problem is?

18:45:20.169 : No start signal detected - forcing start
18:45:20.213 : FIRMWARE_NAME: RepRapFirmware for STM32F4 based Boards FIRMWARE_VERSION: 3.3_2 ELECTRONICS: STM32F4 FIRMWARE_DATE: 2021-06-19 17:14:27
18:45:20.214 : X:0.000 Y:0.000 Z:0.000 E:0.000 E0:-0.0 Count 0 0 0 Machine 0.000 0.000 0.000 Bed comp 0.000
18:45:20.217 : Use P parameter to specify which module to debug
18:45:20.225 : Begin file list
18:45:20.225 : print.g
18:45:20.228 : End file list
18:45:20.229 : Use P parameter to specify which module to debug
18:45:47.759 : WiFi module started
18:45:47.759 : Error: Failed to initialise WiFi module: bad reply format version

I read the previous post on this issue. "One million firmware versions" does not help. Is there a specific way?

Gramm79, 2021-07-02:
...
Recv:19:17:17.714: 80% complete
Recv:19:17:18.855: 85% complete
Recv:19:17:19.995: 90% complete
Recv:19:17:21.151: 95% complete
Recv:19:17:22.188: Upload successful
Recv:19:17:22.689: WiFi module started
Recv:19:17:22.689: Error: Failed to initialise WiFi module: bad reply format version

3dmaksimka, 2021-07-05:
Поставить перемычки как у меня. и прошить прошивкой с этой темы. Мне помогло. другие прошивки почему то не пошли .
[image]