bigtreetech/SKR-2 #36: MULTI_VOLUME support ?

Guilouz, 2021-07-06:
Is it possible to use MULTI_VOLUME feature from Marlin with SKR 2 to use onboard SD card or USB at the same time ?

I have tried many things but not working. Jumpers are placed at the down (without RRF wifi function).

Only SD Card working with this settings (logic) : 

//#define USB_FLASH_DRIVE_SUPPORT
//#define USE_OTG_USB_HOST
#define SDCARD_CONNECTION ONBOARD
//#define MULTI_VOLUME
  #if ENABLED(MULTI_VOLUME)
    #define VOLUME_SD_ONBOARD
    #define VOLUME_USB_FLASH_DRIVE
    #define DEFAULT_VOLUME SV_SD_ONBOARD
    #define DEFAULT_SHARED_VOLUME SV_USB_FLASH_DRIVE
  #endif
  
  
 USB working, SD Card not working with this settings : 
 
#define USB_FLASH_DRIVE_SUPPORT
#define USE_OTG_USB_HOST
#define SDCARD_CONNECTION ONBOARD
#define MULTI_VOLUME
  #if ENABLED(MULTI_VOLUME)
    #define VOLUME_SD_ONBOARD
    #define VOLUME_USB_FLASH_DRIVE
    #define DEFAULT_VOLUME SV_SD_ONBOARD
    #define DEFAULT_SHARED_VOLUME SV_USB_FLASH_DRIVE
  #endif

 USB working, SD Card not working with this settings : 
 
#define USB_FLASH_DRIVE_SUPPORT
#define USE_OTG_USB_HOST
//#define SDCARD_CONNECTION ONBOARD
#define MULTI_VOLUME
  #if ENABLED(MULTI_VOLUME)
    #define VOLUME_SD_ONBOARD
    #define VOLUME_USB_FLASH_DRIVE
    #define DEFAULT_VOLUME SV_SD_ONBOARD
    #define DEFAULT_SHARED_VOLUME SV_USB_FLASH_DRIVE
  #endif

Guilouz, 2021-07-09:
@bigtreetech Any help ?

Teakzieas, 2021-07-26:
When u mean USB working do you mean using a USB flash drive to print G-code.

Guilouz, 2021-07-26:
> When u mean USB working do you mean using a USB flash drive to print G-code?

Yes working for printing.

Teakzieas, 2021-08-05:
I just tested on mine skr2 rev b..MULTI_VOLUME works. What version of marlin are you running?

Guilouz, 2021-08-05:
Latest bugfix. Yes after testing it’s working in Marlin mode with BTT TFT screen but not in touch mode.

Teakzieas, 2021-08-05:
I am getting my BT TFT by this week will try and let you know.

Guilouz, 2021-08-05:
In Touch mode when multi volume is enabled we can only access to usb, we can’t choose sd card or usb drive like in marlin mode.

benek1, 2023-06-23:
#define SDCARD_CONNECTION ONBOARD