bigtreetech/SKR-2 #37: SKR-2 revB USB and power malfunctioning

askmydog, 2021-07-11:
Problem: SKR-2 revB does not power on correctly and does not connect to USB.

Expected behaviors: 
1) SKR-2 revB and BTT TFT35-E3 v3.0 power on when connected to USB without external power.  
2) SKR-2 revB remains powered on after turning on 12V external power supply. 

Observed behaviors:
1) SKR-2 revB does not power on when connected to USB without external power. 
2) No device detected by computer when SKR-2 revB connected via USB
3) SKR-2 revB turns on (main board red LED turns on and Boot screen shows on TFT screen) when connected to external power, but shuts off after approximately 2 seconds (mainboard LED and TFT screen turn off).  Red LED on TFT board remains lit. 
4) If I press the reset button on the main board after powering on the board, the red LED on the SKR-2 mainboard starts flashing at about 2Hz the flashing oscillating between 2 and 5-6 Hz 
5) If I don't disconnect the external power after several seconds when I press the reset button, the main board becomes very hot and the USB port on the computer trips. 
6) If I switch the jumper from VDD to USB, my computer raises a warning that there is a power surge from the USB port, and that the device requires more power than the port can provide. 

Steps to reproduce:  
1) Connected EXP1, EXP2 and RS232 on BTT TFT35 to corresponding pins on SKR-2
2) Connected external power supply from 12v PSU to SKR-2
3) Connected bed heating element to HB and extruder heating element to HE0
4) Powered on to test, appeared to boot normally (device recognized over USB initially)
5) Connected cooling fan to FAN4 (similar to MKS Gen L)
6) Fan ran continuously so turned off PSU, rechecked schematics and moved cooling fan connection to FAN0 and conected extruder fan to FAN4.
7) Turned on external power supply, extruder fan did not turn on, noticed abnormal behavior described above

joshmaceachern, 2021-07-27:
I have a similar issue, I can not verify the TFT behavior as I don't have one but otherwise board behaves the same way.

ypavlovidealo, 2021-11-02:
Same problem here. If any fan is connected, the board will not boot. But it's boot normally after a reset button pressed.

Webaib, 2021-11-02:
Same problem here. If any fan is connected, the board will not boot. But it's boot normally after a reset button pressed.