bigtreetech/SKR-2 #38: Updated Config Request

grizz0blaw, 2021-07-15:
Good evening BTT team,

Can we please get updated configuration.h/configuration_adv.h files (& maybe .bin files for common printer models)? The one listed still shows the old BOARD_BTT_SKR_V2_0 instead of the newer BTT_SKR_V2_0_REV_B that was introduced after the fix to the initial defective boards. I've got an original Ender 5 & I can't get it working at all even trying to use the nightly build. Had to modify the define BOARD_INFO_NAME to BTT_SKR_V2_0_REV_B "pins_BTT_SKR_V2_0_REV_B.h" for it to compile. However, the machine cannot Auto Home without locking up. Thanks in advance for looking into this.

LTSpenny, 2021-07-22:
Hey i just compiled recent marling bug fix for my ender5 plus if you want to give those configurations a shot. Only issue i have is trying to get my hot end fan to be controlled by extruder temp. Let me know and ill post them. Also new to github so ill try to figure out how to submit my sample config.

waxall, 2021-07-22:
Hey thanks but I’m on Klipper so I don’t think the marlin configs would work?

inbetweendreams, 2021-07-23:
> Hey i just compiled recent marling bug fix for my ender5 plus if you want to give those configurations a shot. Only issue i have is trying to get my hot end fan to be controlled by extruder temp. [...]

 Can I get that from you? I compiled one for my ender 5 plus but haven't tried it because I have update the board to rev B (they sent me the part, I hope this weekend) I would like to compare it as I am new to building my own. Ender5plus SKR2 Mosquito Magnum Bondtech LGX.