bigtreetech/SKR-2 #43: SKR 2 rev B with BTT TMC 2208 v3 -> TMC CONNECTION ERROR

howardjones, 2021-07-25:
I have a new SKR 2 rev B board and 5 BTT TMC2208 v3 stepsticks - I am getting "TMC CONNECTION ERROR" from the display at boot up, and M122 just shows:

    Testing X connection... Error: All LOW
    Testing Y connection... Error: All LOW
    Testing Z connection... Error: All LOW
    Testing E connection... Error: All LOW

Nothing else is connected currently, apart from the BTT 12864 RGB mini display, and power/serial from USB. The jumper is set underneath each socket to enable UATRT. As far as I can see, the TMC2208 modules already have a link on the UART config jumper J2 (it's really tiny, but I think it is there).  Link is between the bottom two pins.

Marlin 2.0.9.1 is configured for TMC2208 on all axes.

What else is needed?

Guilouz, 2021-07-25:
TMC are not working in USB mode, it's normal.

VanessaE, 2021-09-02:
https://github.com/MarlinFirmware/Marlin/issues/22689