bigtreetech/SKR-2 #44: Dual Z endstop configuration problem

eccentrichead, 2021-07-26:
#### Issue:
When Configuring of Dual Z endstop in SKR2 board the Z2_min fails to get mapped to XMAX or ZMAX pins. 

#### Steps to reproduce: 

* Download the latest Marlin2.x release from main stream and setup the dual z option:
	- `config.h` file setup:
	```
	   #define USE_XMIN_PLUG 
	   #define USE_XMAX_PLUG //[Ai3M - dual z-endstops]
	   #define USE_YMIN_PLUG
	   #define USE_YMAX_PLUG
	   #define USE_ZMIN_PLUG
       #define USE_ZMAX_PLUG
       
       #define X_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop. 
       #define X_MAX_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop. 
       #define Y_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop. 
       #define Y_MAX_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop. 
       #define Z_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop. 
       #define Z_MAX_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop. 
       
       #define X_DRIVER_TYPE DRV8825 
       #define Y_DRIVER_TYPE DRV8825
       #define Z_DRIVER_TYPE DRV8825 
       #define E0_DRIVER_TYPE A4988
       #define Z2_DRIVER_TYPE DRV8825 
       
       #define X_HOME_DIR -1
       #define Y_HOME_DIR -1
       #define Z_HOME_DIR -1
     ```

	* `config_adv.h`:
	```
	//
	// For Z set the number of stepper drivers
	//
	#define NUM_Z_STEPPER_DRIVERS 2   // (1-4) Z options change based on how many
	#if NUM_Z_STEPPER_DRIVERS > 1 
   // Enable if Z motor direction signals are the opposite of Z1
  //#define INVERT_Z2_VS_Z_DIR
  //#define INVERT_Z3_VS_Z_DIR
  //#define INVERT_Z4_VS_Z_DIR

  #define Z_MULTI_ENDSTOPS  //[skr2]
  #if ENABLED(Z_MULTI_ENDSTOPS)
    #define Z2_USE_ENDSTOP          _XMAX_
    #define Z2_ENDSTOP_ADJUSTMENT   0
    #if NUM_Z_STEPPER_DRIVERS >= 3
      #define Z3_USE_ENDSTOP        _YMAX_
      #define Z3_ENDSTOP_ADJUSTMENT 0
    #endif
    #if NUM_Z_STEPPER_DRIVERS >= 4
      #define Z4_USE_ENDSTOP        _ZMAX_
      #define Z4_ENDSTOP_ADJUSTMENT 0
    #endif
  #endif
  #endif 
   ```

#### Actual result:
* Issuing `m119` reveals the `z_max` port getting picked up instead of `z2_min`port:
```
>>> m119
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
z_max: open
```

* Changing the `_XMAX_` to `_ZMAX_` instead in the `z2` endstop config, results in correct port mapping:
	- Config_adv.h:
```
  #define Z_MULTI_ENDSTOPS  //[skr2]
  #if ENABLED(Z_MULTI_ENDSTOPS)
    #define Z2_USE_ENDSTOP         _ZMAX_ 
    #define Z2_ENDSTOP_ADJUSTMENT   0
    #if NUM_Z_STEPPER_DRIVERS >= 3
      #define Z3_USE_ENDSTOP        _YMAX_
      #define Z3_ENDSTOP_ADJUSTMENT 0
    #endif
    #if NUM_Z_STEPPER_DRIVERS >= 4
      #define Z4_USE_ENDSTOP        _ZMAX_
      #define Z4_ENDSTOP_ADJUSTMENT 0
    #endif
  #endif
#endif
```
* `m119` result:
```
>>> m119
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
z2_min: open
```

#### Expected result:
* The `XMAX` or `YMAX` port should get mapped as `z2_min`port.

#### Additional information
* The same configuration works perfectly without issue in an SKR v1.3 board using the same marlin version.

eccentrichead, 2021-07-29:
@bigtreetech , kindly have a look. Thanks!

SMTDesigns, 2021-09-22:
pins_BTT_SKR_V2.0_common.h defines what hardware pin on the board gets "attached" to the main marlin software.
Line 166 is
#elif ENABLED(Z_MULTI_ENDSTOPS)
  #ifndef Z_MIN_PIN
    #define Z_MIN_PIN                       PC0   // Z-STOP
  #endif
  #ifndef Z_MAX_PIN
    #define Z_MAX_PIN                       PC15  // PWRDET
  #endif
#else
  #ifndef Z_STOP_PIN
    #define Z_STOP_PIN                      PC0   // Z-STOP
  #endif
#endif

I was trying to use the I_MIN assignment because there is no MAX definitions in the latest code I downloaded a few days ago. I ended up seeing this thread and tracked down where Z_MAX_PIN was defined.

SMTDesigns, 2021-12-16:
I moved on to the MKS Robin Nano V3.0 board that is working great. The BTT ended up having issues after I got one working (don't remember what now...). I ended up changing to the MKS Robin Nano v3.0 board and it is working great on two printers so far.