bigtreetech/SKR-2 #47: SKR 2 Homing Problems?

scaramonga, 2021-07-30:
Hello I wonder if anyone can help me?

Ender 3 V2 and fitted an SKR 2 board/2209 steppers with TFT35 E3 v3 - BLTouch & Dual Z Axis (each on their own stepper)

2.0.9.1 Marlin compiled by myself.

The problem I'm having, is all the endstops seem to be getting ignored?  A M119 shows they are working OK, but doing an 'Auto Home' results in all the axis grinding when they reach these points?, like it's trying to continue past the endstop marks?  I have sensorless homing disabled.

Could someone maybe check my configs to see if I have missed something?  I'm at my wits end!

One thing I did notice, is that with M119 command, it shows XYZ triggered when I test when pressing them, open when released, but it throws up a continuous error of 'Couldn't X:0 Y:0 Z:0'?

I'd be grateful for any help guys.  Thank you.

scaramonga, 2021-07-30:
Done with this board.  Worst ever from the series!  It's got faults galore, and one's back is up against the wall when the 'actual' hardware is not good.  Take it off the market please BTT.

SciFiModelmaker, 2021-08-03:
When you say sensorless homing is disabled, did you mean by means of _not_ installing the diag pin jumper, or in Marlin, or both? I've got one of these and I'm going to try to get it working with EZABL.

scaramonga, 2021-08-03:
Both! :)  Good luck with EZABL, you will need it ;)

scaramonga, 2021-08-04:
Well I have to apologize.  It's nothing to do with the SKR 2, in fact, I really do not know what it's to do with?  I actually give up!  Got me stumped?  I have tried everything.  Endstop Pullups/Downs disable/enable, endstop reverse/enable/disable, with probe, without, minimum Marlin config, enabling/disabling not a lot, running board with just steppers, it's all no go on the Ender 3 V2 here, and that is with the SKR 2 & 1.4 Turbo, each exactly the same?  Replaced the stepper motors for brand new ones, exactly the same, replaced the stepper wires, replaced the endstops, exactly the same?  G28 from a terminal, results in the bed moving forward till it reaches the end, but tries to keep on going with an awful grinding, and nothing else happens, so I have to pull the plug.  Motors all correct, as I can move them in testing (an inch at a time), XYZ, but as soon as you let it go, all hell breaks loose?  M119 tests OK for all connections.

I'm beginning to think the TFT35?, it's it's own system, so what else is there?

SciFiModelmaker, 2021-08-05:
@scaramonga I too have had experience with the 1.4 Turbo, and when it works it's great, but my heavily-modded Ender-3 is offline more than it's on. I was beginning to think I was just randomly frying bed MOSFETs, but learned something while testing them.

The threshold voltage from the MOSFET spec sheet is 3 volts, so I set up a buck converter on a breadboard at 3.2v and began testing, and found what I expected to find; MOSFETs that wouldn't switch on. The power supply was janky and I wanted something more robust for testing multiple boards, so I broke out the bench supply, and set it on a preset at 3.5v. 3v MOSFETs that weren't working at 3.2v were now operating as expected at 3.5v.

This got me thinking; What if powering my Pi from the accessory 24v on the SKR 1.4 Turbo was causing the MOSFETs to have insufficient voltage to switch on? I'm next planning to directly measure what's happening on the board to check. I may just test it empirically first, by isolating power for the Pi and trying a 1.4 Turbo whose MOSFETs pass at 3.5v but fail at 3.2.

My point is this; I never make a mistake until I know what I'm doing.

I like where you're going with the TFT35. Eliminating variables can help when stuck on a problem. Have you tried it with only EXP1/2 and in Marlin mode, not using the TFT cable, or some variation of that? I ask because the SKR-2 manual only references the TFT connector when discussing connection to a Pi.

scaramonga, 2021-08-06:
Had a play about with it again today, and I dunno if this is relevant?  But I decided to set my far left front home position by hand, and note the coordinates on screen (through Pronterface) connected to lappy, and here they are: (5mm from front left and 5mm from forward, and nozzle 5mm from bed):

X -13.00 Y +164.20 Z -48.00  (bed size 230x230)

Something is off there?  I did try Marlin mode without the TFT cable, exactly the same, no difference.  It's like the printer does not know the area it's in?  I dread selecting 'home', lol, just to hear the grinding of they motors now, as it tries to go to infinity.

New printer coming next week, as I have given up with this now lol.  Gonna cut my losses and send both BTT boards back, but yes, I will still lose sleep over it, as it's annoying me :)

SciFiModelmaker, 2021-08-07:
I'm apparently just cursed with BTT boards. Fried X,Y,Z TMC2209, no idea how. I just connected everything, sensorless homing, DIAG jumpers in place, fired it up. No problem heating bed or hotend, tried to move an axis, no joy. Powered down and monkeyed around a bit to try and diagnose moving issue. Next boot, I get ALL LOW and M122 reports no comms with X, Y or Z.

Wish I could give up. It's my only FDM and I'm too far down the rabbit hole to go back now. Best of luck with your new machine.

scaramonga, 2021-08-08:
Yes, 'ALL LOW' is exactly what I was getting also, which I think is to do with stepper drivers?, but I had a set of 2208's here also, so updated firmware and tried them also, still the same?

I really think the BTT boards are a 'hit & miss', as to whether you get one that works properly?.  Poorly made, with components that do not function as they should, or only do so outwit the specs.  Never will I get one again.

slf495, 2021-08-10:
I am having a similar problem after just getting this board (actually 2nd) the first board was DOA), when I got so send it home, the axis move a few mm's and stop. I am ordering new tmc 2209's as the ones I had prior i had to clip a pin off, and wondering if that's the issue? any ideas?
    Also, getting an error .......    (i'm including the format the error is on the screen, JUST in case it matters)
"failed to enable bed lev
eling   echo: bed leveling 
OFF echo: fade height 1 
0.00     ok
   Anyway, I would greatly appreciate any help you can dend my way, as it is i'm not happy with this board either so far

SciFiModelmaker, 2021-08-10:
@slf495 , those are normal messages related to bed leveling. 'Failed to enable bed leveling' is because there is no valid mesh from ABL stored in EEPROM. This will go away after you've leveled using ABL and a mesh is stored. 'Fade height 10' is just the printer announcing the default fade height for ABL. Fade height is the height at which all the warps and bends your ABL finds on your bed surface are finally made 'level'. 'ok' means, well, OK. 

Is it operating normally otherwise?

SciFiModelmaker, 2021-08-10:
@scaramonga SKR-2 is going back to Amazon.  Meanwhile, I've resurrected one of my 1.4Turbos after discovering it was indeed a low voltage condition that was causing heating failures. I was powering my PI off the 24v accessory and a buck converter. This was causing the logic voltage to drop just below the 'on' threshold voltage for the MOSFET, preventing it from switching on. I split my 24v before the board, and all is well. For now. 

Best of luck, and I'm jealous over the Ender 6. Always wanted to do Core XY.

slf495, 2021-08-10:
Ok, so that’s the problem!!!

Theming to hone the z, where the bltouch USED to be at the center for homing, it’s not almost fully forward, but now centered (no idea what I did, but it’s closer). 
Trying to abl, it’s going from back right forward, actually going sideways for whatever reason, so I can’t complete the level

Can you please send what changes I need for all the axis in marlin, the bltouch, and filament runout sensor, the fan for the heat block, and the cooler fan. I am at my wits end and am sick of going back and forth at this rate every change doing little or nothing and I’m about to bin the damn thing (as the first board to begin with was dead on arrival)

slf495, 2021-08-10:
It’s an ender 5 pro, I don’t know if that was mentioned anywhere

Sack87, 2021-08-15:
I just got my SKR 2 board working with all of the above as per original post.

CHECK YOUR ENDSTOP PINS ARE CORRECT! My pins were incorrect when switching from 1.4 turbo to 2. "C" marked on the endstop is the signal, so PE whatever axis it is on the board. "NO" is the ground and "NC" is the 5v.
I have a 3DTouch and just had to plug the black and white wire  connector opposite to suit the board pinouts, the 3 pin is correct for the board pinout.

slf495, 2021-08-15:
I managed to get it working thanks, was curious if your having any issues with the board overheating? I’m trying to do the thing from teaching tech and using the 2 wires connection the 3rd pin to the ground on two of the accessory ports, dunno if it’ll work yet, and disabling the safe_power_protect line in the adv config. I should have the B board where this shouldn’t be an issue so I’m not sure what’s wrong