bigtreetech/SKR-2 #49: Z homing issue

bak218, 2021-07-30:
I am having a problem with z homing. I can home to X and Y but when I try to home Z the Z axis does not lower the BLtouch deploys and stores correctly. I can raise and lower Z with the movie buttons, but none of the Z homing  or bed leveling options work. 

SKR 2 motherboard with Ender 3 pro and TFT35 E3 V3.0 display

[Configuration_adv.zip](https://github.com/bigtreetech/SKR-2/files/6910053/Configuration_adv.zip)

scaramonga, 2021-07-30:
Good luck :)

bak218, 2021-07-31:
> Good luck :)

I saw your previous post. I'm think of going to the creality 4.2.7 board and display. But my goal is dual extruder. OEM is probably simpler.

scaramonga, 2021-07-31:
Yeah, I have given up with this board.  So many faults it's unreal, and there is a reason that there is no support for it, I'm talking support like, for the 1.4/1.4 Turbo.  This board doesn't seem to exist on the internet?, apart from a handful of 'basic' videos that tell you nothing.  It's a sorry state of a board from BTT, which I suspect has more problems than the initial mosfet disaster.  Never have I had a 3D Printer board with so many problems, and even Marlin, nor anyone out there, has 'configs' for it.  It's a nightmare, and quite frankly, not worth anyone's time.  I have spent a good few hours in VS Studio Code trying to compile something that is going to work, but never got there yet.

scaramonga, 2021-08-01:
BTW.  BLTouch wiring is completely 'reversed' on this board, as in 'completely'!  ;)  What was left to right, now goes right to left :), if your using Creality 3.1 probe.

bak218, 2021-08-01:
That's awesome. I ordered the skr E3 turbo it's supposed for strictly for the Ender 3 and they have firmware for it. I was fighting with the  Skr 2 board for several weeks.