bigtreetech/SKR-2 #50: TMC connection error - TMC 2130

umangmd, 2021-07-31:
I was wondering if someone could help me figure this one out. I'm new to setting up a board with new drivers so it's likely i've made a mistake somewhere. Currently using an SKR 2.0 rev B board with TMC 2130 drivers but Marlin always says there's a TMC connection error. I pretty sure i've done all the hardware connections correctly (picture attached). I hoping it's something in the firmware if anyone can help me out, i've attached the config files. Thanks
[marlin_config.zip](https://github.com/bigtreetech/SKR-2/files/6911185/marlin_config.zip)
[image]

SciFiModelmaker, 2021-08-10:
@umangmd My SKR-2 was throwing ALL LOW after installation and first movement tests failed. It fried the TMC2209's on any axis I tried to move. I did not try to move the extruder. That driver is fine.

Eds100, 2022-02-11:
Same here The upgraded SKR 2 with TMC2130 and I also get this " TMC connection error" has anyone come up with an answer.

kelvinmead, 2024-06-21:
2 more years later. im thinking this is a dead combination!