bigtreetech/SKR-2 #51: Ender 3v2 DWIN LCD Not Working

Seminole81, 2021-08-03:
Hi, I am aware there are issues with this, which seem to be linked to the DWIN LCD screen.

Can this be fixed similarly to the Mini? Seems like Marlin DEVs already coded in a fix, but just not for the SKR 2? Will we need to create a new jumper ribbon as well similar to the fix?

https://github.com/MarlinFirmware/Marlin/pull/22288

Sorry, a bit new to all of this still.

McSneaky, 2021-08-07:
I have same issues. I've tried with similar jumper ribbon, but I can't get it to compile when I enable `DWIN_CREALITY_LCD`

Seminole81, 2021-08-07:
> I have same issues. I've tried with similar jumper ribbon, but I can't get it to compile when I enable `DWIN_CREALITY_LCD`
> 
> Tried to follow this guide: https://www.youtube.com/watch?v=gzrw0lABCBE
> And then the mini one and combination of them both, but doesn't seem to work :|

Yeah, I found the same one, but couldn't follow it with my lack of Hungarian ha. I did try similar though with the 1.4 and even Mini E3 etc. Seems a jumper cable needs made and pinouts. I ended up buying a TFT35 display and got that working dead quick, but I actually prefer the DWIN...

KBKreisler, 2021-09-29:
I am able to get DWIN to work - I had to add 'BTT_SKR_V2_0_REV_B' TO THE INCLUSION STATEMENT
 @ LINE 507 Marlin\src\inc\Conditionals_LCD.h

 #if MB(BTT_SKR_MINI_E3_V1_0, BTT_SKR_MINI_E3_V1_2, BTT_SKR_MINI_E3_V2_0, BTT_SKR_E3_TURBO, BTT_SKR_V2_0_REV_B )

as well as include an additional else-if statement with board specific pin mapping @ Line  483 in Marlin\src\pins\stm32f4\pins_BTT_SKR_V2_0_common.h

#elif ENABLED(HAS_DWIN_E3V2)
	// Added Support for DWIN_CREALITY_LCD - KBK Edit
  #define BTN_EN1                  EXP1_03_PIN
  #define BTN_EN2                  EXP1_04_PIN
  #define BEEPER_PIN               EXP1_05_PIN
  #define BTN_ENC                  EXP1_06_PIN

#endif // HAS_WIRED_LCD

pinout matches, but the Tx / Rx are split off to terminate on the TFT UART ports

                      -----                                         -----                                          ---------------
                5V  | . . | GND                             5V | . . | GND                               |  .   .   RX   TX  .  |                                 
       BTN_EN1 | . . | BTN_EN2            BTN_EN1 | . . | BTN_EN2                          ---------------
  BEEPER_PIN | . . |  BTN_ENC       BEEPER_PIN| . . |  BTN_ENC                               TFT
              TX    | . . |   RX                                 -  | . . |  -
                  -   | . . |  -                                    -  | . . |  -
                      -----                                          ----- 
                     DWIN                                        EXP1

I am having some icon issues in the latest bug fix, but the stable release functions fine.
Icon issue : Icons only show up as text, and icon halos remain around the icon objects the main screen after the curser has moved off. This resets when returning from a menu.

Submenus and functions all seem to work fine though

KBKreisler, 2021-10-15:
[BIGTREETECH SKR 2-Pin.pdf](https://github.com/bigtreetech/SKR-2/files/7353833/BIGTREETECH.SKR.2-Pin.pdf)
Full pinout of board

sagicastro, 2021-10-16:
> [image] On the TFT port PA9/PA10

@KBKreisler Thanks! works fine, also like yours without icons

LightningShark25, 2021-10-22:
Thanks so much for this post.   I'm up and running.
Like others have said, I don't see the icons/symbols and there are a few glitches.  But for the most part its working well.  

I have had trouble in that I tried putting your pin mapping in the #if'HAS_WIRED_LCD'.  I can see from Intellisense and from confirming my build in the board, that this 'HAS_WIRED_LCD' is not active.

Tried searching for the path that didn't set it, but gave up.

I worked around it by 'putting the definition outside the IF - which will no doubt catch me out down the track!

If anyone knows what I might have done wrong - be keen to hear!  Thanks

cjmalchow, 2021-12-11:
> 

Hi, I see that after ENABLED you listed the display as HAS_DWIN_E3V2. In the base config for Ender 3 v2 it lists the display as DWIN_CREALITY_LCD in configuration.h . Should I still use HAS_DWIN_E3V2?

Thanks!

Chris-Jayden, 2022-03-04:
My DWIN screen just flickers on boot; is there something I'm missing here?
(I'm attempting to use Jyer's version of the Marlin 2.0.X Bugfix branch)

For "Conditionals_LCD.h", I've added "BTT_SKR_V2_0_REV_B" to the MB statement:
`// Aliases for LCD features
#if EITHER(DWIN_CREALITY_LCD, DWIN_CREALITY_LCD_ENHANCED)
  #define HAS_DWIN_E3V2_BASIC 1
#endif
#if EITHER(HAS_DWIN_E3V2_BASIC, DWIN_CREALITY_LCD_JYERSUI)
  #define HAS_DWIN_E3V2 1
#endif

// E3V2 extras
#if HAS_DWIN_E3V2 || IS_DWIN_MARLINUI
  #define SERIAL_CATCHALL 0
  #ifndef LCD_SERIAL_PORT
    #if MB(BTT_SKR_MINI_E3_V1_0, BTT_SKR_MINI_E3_V1_2, BTT_SKR_MINI_E3_V2_0, BTT_SKR_MINI_E3_V3_0, BTT_SKR_E3_TURBO, BTT_SKR_V2_0_REV_B)
      #define LCD_SERIAL_PORT 1 //Default 1`

For "pins_BTT_SKR_V2_0_common.h", I've tried exclusively either of:

```
    #define BTN_EN1 PE13 //EXP1_03_PIN
    #define BTN_EN2 PE12 //EXP1_04_PIN
    #define BEEPER_PIN PE11 //EXP1_05_PIN
    #define BTN_ENC PE10 //EXP1_06_PIN

OR

    //#define BTN_EN1               EXP1_03_PIN
    //#define BTN_EN2               EXP1_04_PIN
    //#define BEEPER_PIN           EXP1_05_PIN
    //#define BTN_ENC               EXP1_06_PIN
```

I get a successful compile with no yellow or red errors, but flashing the board and connecting my modified ribbon to EXP1 and split-off orange and yellow wires to TFT_03 (PA9, RX) and TFT_04 (PA10, TX). The DWIN screen just flickers between an off state and a 0 brightness value on state (it's slightly brighter than the off state).

I don't need to do anything extra like add a define argument for PA9 and PA10, do I?

Also, I've seen this mentioned in the Mini E3 fix, but I'm not sure it applies to the SKR 2.0:
```
#define SERIAL_PORT -1
#define BAUDRATE 115200

//#define SERIAL_PORT_2 -1
//#define BAUDRATE_2 250000
```

Any suggestions on how to proceed?

pmead7399, 2022-03-29:
Oh man I could really use some help,,, I've been on this for 2 weeks and this thread looks very promising. @[KBKreisler](https://github.com/KBKreisler) you wrote:  

as well as include an additional else-if statement with board specific pin mapping @ Line 483 in Marlin\src\pins\stm32f4\pins_BTT_SKR_V2_0_common.h

#elif ENABLED(HAS_DWIN_E3V2)
// Added Support for DWIN_CREALITY_LCD - KBK Edit
#define BTN_EN1 EXP1_03_PIN
#define BTN_EN2 EXP1_04_PIN
#define BEEPER_PIN EXP1_05_PIN
#define BTN_ENC EXP1_06_PIN

#endif // HAS_WIRED_LCD

the first edit you stated wasn't found on line 507 and I suspect the line 483  has changed also and I don't know exactly where to stick that in. If there is any way you (or anyone else here) could help me with that I would be very thankful.

I have firmware that compiles, not tested quite yet, and I'm sure I could Pronterface or Octopi it but I need bltouch so I can have the 2nd z-axis stuff set up and you need the screen for bltouch and you need bltouch.... blah blah blah.

Thanks! I'm crossing my fingers you get this.

P.S. if it matters I'm on Ender 3 V2 (just installed extender kit) and SKR-2_429