bigtreetech/SKR-2 #52: THIS BOARD IS A P.O.S. HOMING HELP REQUIRED

Sack87, 2021-08-07:
I built up my own A10M/CR-10 S4 hybrid printer using this board, previously I had the A10M with SKR 1.4 turbo running with TFT35

Basically all I did was update my old 100% working 1.4 turbo firmware and change the bed size to suit the CR-10 S4 size and the board to SKR 2, adding in the BL touch.

PROBLEMS:
X axis wont move to home when selected from TFT screen
Y axis won't move to home when selected from TFT screen
Z axis will not move to centre of bed to home
Z safe homing will absolutely not work no matter what I try (used offsets, centre bed etc)
ABL just probes the first spot over and over and over and over and over and over
It won't verify end stops even though commented, it just hits them and stays hard against them.
It never returns to 0,0 after homing 

I've been fucking around with the firmware for like 2 weeks, following all of the guides I can find on the internet/youtube and nothing works. Am I just wasting my time with this board?

SciFiModelmaker, 2021-08-10:
@Sack87 I'm sending my SKR-2 back, and I'm not the type of person who gives up easily. Simple configuration, like yours, from a known-good 1.4Turbo. Installed TMC2209's, set up stallguard / sensorless homing, set the DIAG jumpers per the manual, flashed it. Checked the bed and hotend, both working fine. Start moving axes, and that's when the fun stopped. I got unpredictable responses (wrong direction mostly), followed by some low growling from the steppers motor(s). That freaked me out so I scrammed it. On restart it gave the ALL LOW error that indicates your trinamic stepper drivers aren't communicating correctly. All three TMC2209, X,Y and Z are fried, confirmed on a known good 1.4 Turbo.

I've gone back to the 1.4 Turbo. I recommend you do the same if you value your sanity.

Sack87, 2021-08-14:
Well I finally fucking succeeded in getting my A10Mega flashed and working on the SKR 2 board!!! Excuse the language but it's been a frustrating month trying to get it to work.

I'll post my config files tomorrow but here is a run down:
A10M that has been upsized to 400x400x400
Closed loop steppers on X and Y axis (this means NO SENSORLESS HOMING as they don't support it)
TMC2209 drivers
3DTouch that actually homes to exact centre
Z safe homing works as does ABL
1000W AC powered silicone heat bed
TFT35 fully functional but manual move buttons on X/Y axis are inverted (not fussed though)