bigtreetech/SKR-2 #56: upgrading Anycubic I3 Mega with SKR-2 - Power input amperage

RyantHults, 2021-08-15:
I am attempting to upgrade my Anycubic I3 Mega from a Trigorilla 1.1 to the SKR-2, and have a question about which amperage to use. The Trigorilla 1.1 has inputs from the power supply for 5A and 11A. I'm assuming I'll only need one of these, but I'm not sure which one should be used? What amperage does the SKR-2 expect?

adcgn, 2021-08-18:
Just use the 5A for the Power/GND and the 11A for the bed. You need to connect both.

RyantHults, 2021-08-30:
from looking at other tutorials and builds, it appears that either one can be used, and the un-used one should be disconnected from the power supply. The board will draw the however much amperage it needs (up to the power supply's limits).

closed 2021-08-30 by RyantHults