bigtreetech/SKR-2 #58: Error: open failed in procedure 'program' ** OpenOCD init failed ** shutdown command invoked

stweed69, 2021-08-19:
i have bought two skr 2 boards both not working because sd card port not working have also tried btt marlin listed here and vanilla marlin code compiles fine but wont upload  and both board sd card slot not working... uploading via terminal no go... this is what i get.....

Processing BIGTREE_SKR_2 (platform: ststm32@~12.1; board: marlin_STM32F407VGT6_CCM; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_STM32F407VGT6_CCM.html
PLATFORM: ST STM32 (12.1.1) > STM32F407VG (128k RAM, 64k CCM RAM, 1024k Flash
HARDWARE: STM32F407VGT6 168MHz, 128KB RAM, 1MB Flash
DEBUG: Current (stlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 0.0.0
 - framework-cmsis 2.50501.200527 (5.5.1)
 - tool-dfuutil 1.9.200310
 - tool-openocd 2.1000.200630 (10.0)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TMCStepper> 0.7.1       
|   |-- <SoftwareSerial> 1.0 
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.8.5
|-- <U8glib-HAL> 0.5.0       
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <SoftwareSerial> 1.0     
|-- <IWatchdog> 1.0.0        
|-- <Wire> 1.0
Building in release mode
Checking size .pio\build\BIGTREE_SKR_2\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.1% (used 23660 bytes from 131072 bytes)
Flash: [==        ]  21.5% (used 225068 bytes from 1048576 bytes)
Configuring upload protocol...
AVAILABLE: dfu, jlink, stlink 
CURRENT: upload_protocol = stlink
Uploading .pio\build\BIGTREE_SKR_2\firmware.bin
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

hla_swd
none separate

Error: open failed
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

*** [upload] Error 1

stweed69, 2021-08-19:
i think i have found the problem btt has set the board to only except sdcard flashing u cant flash via terminal.... well that makes both boards useless because both boards i have ordered  have arrived with non-functioning sdcard ports  very shoddy workmanship btt  very sad.

ellensp, 2021-12-17:
"CURRENT: upload_protocol = stlink"
Ie you need a stlink hardware device connected to the board to use this upload method