bigtreetech/SKR-2 #63: SKR2 Burning up TMC2208 Drivers in UART mode

joaustin, 2021-08-30:
I have an SKR 2 Rev B board and it is buring up TMC2208 stepper drivers. It is consistently the E Motor and the Z2 Motor. Help!

Send: M122 S0
Recv: 		X	Y	Z	Z2	E
Recv: Enabled		false	false	true	false	false
Recv: Set current	650	650	650	650	800
Recv: RMS current	642	642	642	1160	1436
Recv: MAX current	905	905	905	1636	2025
Recv: Run current	20/31	20/31	20/31	20/31	25/31
Recv: Hold current	10/31	10/31	10/31	10/31	12/31
Recv: CS actual	10/31	10/31	10/31	0/31	0/31
Recv: PWM scale
Recv: vsense		1=.18	1=.18	1=.18	0=.325	0=.325
Recv: stealthChop	true	true	true	false	false
Recv: msteps		16	16	16	256	256
Recv: interp		true	true	true	false	false
Recv: tstep		max	max	max	0	0
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false	false
Recv: triggered
Recv:  OTP		false	false	false	false	false
Recv: pwm scale sum	12	12	37	0	0
Recv: pwm scale auto	0	0	3	0	0
Recv: pwm offset auto	36	36	101	0	0
Recv: pwm grad auto	14	14	14	0	0
Recv: off time	4	4	4	0	0
Recv: blank time	24	24	24	16	16
Recv: hysteresis
Recv:  -end		2	2	2	-3	-3
Recv:  -start		1	1	1	1	1
Recv: Stallguard thrs
Recv: uStep count	8	8	8	0	0
Recv: DRVSTATUS	X	Y	Z	Z2	E
Recv: sg_result
Recv: stst					*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:0A:00:00
Recv: 		Y	0xC0:0A:00:00
Recv: 		Z	0xC0:0A:00:00
Recv: 		Z2	0x00:00:00:00	 Bad response!
Recv: 		E	0x00:00:00:00	 Bad response!
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing Z2 connection... Error: All LOW
Recv: Testing E connection... Error: All LOW
Recv: ok

VanessaE, 2021-09-02:
Yep, I'm another victim of this.  The most likely problem is that backwards safe power protect feature.  It kills TMC2208's even on an SKR 2 Revision B.  Lost two older Eryone drivers and a BTT v3.0 driver before I realized what was going on.

Look in `Configuration_adv.h` for`DISABLE_DRIVER_SAFE_POWER_PROTECT` (near line 3580 at present). If that line is commented-out, un-comment it, save, and recompile.

gzalo, 2021-09-04:
Ouch! Did you contact BigTreeTech about this? It sounds bad. 
I have some questions for both of you, if you could answer them (or as many as possible) would be great to try and determine what is going on:
- Are the drivers getting burned as soon as you plug it in? or after a while?
- Were the motors connected to the board or not?
- Were you powering the board through USB, external voltage or both when they got burned? if both, in which order did they startup?
- What was the position of the J17 DC SEL jumper: USB or VDD?
- What was the configuration of the jumpers that are below the drivers?

VanessaE, 2021-09-04:
> Are the drivers getting burned as soon as you plug it in? or after a while?

Slightly warm from the get-go, and only got hot if I tried to move the respective motor, for as long as the driver was turned on (`M84` would turn it off and it would start to cool again).

When I say hot, I mean hot enough to burn if I left my finger on the heat sink, and it would get there within a few tens of seconds.  Obviously way hotter than the driver would normally be.

> Were the motors connected to the board or not?

Most of the time, the motors were connected (I was trying to do routine motion tests, with intent to print something shortly thereafter), but the drivers would also heat up without a motor.  Probably by that point, the damage was already done.

> Were you powering the board through USB, external voltage or both when they got burned? if both, in which order did they startup?
> What was the position of the J17 DC SEL jumper: USB or VDD?

Naturally with the 24v supply connected, and the jumper set to VDD.  TMC drivers don't initialize and make UART comms available without it. USB was connected for normal command/control/etc.  Since that jumper controls power to the motherboard's 5v rail, it (and 3.3v) would come up as 24v does.

> What was the configuration of the jumpers that are below the drivers?

One shunt apiece, on the UART pins.

I set up a return/refund of the defective board and bought a new one along with a handful of new TMC2208's.

Here is my configuration that I'll be using on the new hardware:

[Marlin-configs-20210902.zip](https://github.com/MarlinFirmware/Marlin/files/7102643/Marlin-configs-20210902.zip)

<details><summary>And here's the g-code I'll use to set up the EEPROM</summary>

```gcode
  G21    ; Units in mm (mm)
  M149 C ; Units in Celsius
; Filament settings: Disabled
  M200 S0 D1.74
; Steps per unit:
 M92 X80.00 Y80.00 Z400.00 E885.00
; Maximum feedrates (units/s):
  M203 X500.00 Y500.00 Z200.00 E150.00
; Maximum Acceleration (units/s2):
  M201 X2000.00 Y2000.00 Z250.00 E1500.00
; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
  M204 P1500.00 R1000.00 T5000.00
; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
  M205 B20000.00 S0.00 T0.00 J0.05
; Home offset:
  M206 X0.00 Y0.00 Z0.00
; Material heatup parameters:
  M145 S0 H180.00 B65.00 F0
  M145 S1 H215.00 B80.00 F0
; PID settings:
  M301 P23.61 I2.31 D60.30
  M304 P20.00 I3.00 D300.00
; User thermistors:
  M305 P0 R4700.0 T93520.0 B5071.0 C0.000000160 ; HOTEND 0
; Retract: S<length> F<units/m> Z<lift>
  M207 S3.00 W0.00 F3000.00 Z0.00
; Recover: S<length> F<units/m>
  M208 S0.00 W0.00 F3000.00
; Auto-Retract: S=0 to disable, 1 to interpret E-only moves as retract/recover
  M209 S0
; Z-Probe Offset (mm):
  M851 X-22.00 Y0.00 Z-0.4
; Stepper driver current:
  M906 X1000 Y1000
; Hybrid Threshold:
  M913 X100 Y100
; Driver stepping mode:
; Filament load/unload lengths:
  M603 L500.00 U700.00
  M503

; (plus I'll have to re-probe for bed leveling)
```
</details>

Let me know if I missed something on those?

joaustin, 2021-09-04:
I have not tried Vanessa's suggestion yet as I am out of functional drivers.
> Ouch! Did you contact BigTreeTech about this? It sounds bad.
Yes - I have a case opened with them. I have been going back and forth and it seems they are going to replace the board and steppers. We will see how long that takes
> I have some questions for both of you, if you could answer them (or as many as possible) would be great to try and determine what is going on:
> 
> * Are the drivers getting burned as soon as you plug it in? or after a while?
They will work for a while - when they are first plugged in for the first couple power cycles and then they will start erroring out on intial power up (fixable with an immediate power cycle) and then they will just completely stop working
> * Were the motors connected to the board or not?
Yes motors are connected and functional
> * Were you powering the board through USB, external voltage or both when they got burned? if both, in which order did they startup?
24V power supply only
> * What was the position of the J17 DC SEL jumper: USB or VDD?
VDD
> * What was the configuration of the jumpers that are below the drivers?
UART pin only jumpered (see below XX indicates two jumpered pins
I I I 
I I I
XXI

I I I

VanessaE, 2021-09-08:
So, I got my replacement board, another SKR 2 Rev. B (I checked -- Q1 is HY1904),  and the TMC2208's I bought all came pre-configured for UART.

Set the jumpers, compiled, flashed, then powered-off, added my drivers, and powered-on.  No go -- exact same problem as on the first board, i.e. "TMC connection error" on the display (and if I plug in USB, `M122` will show "ALL LOW" for one or more drivers), and drivers begin warming-up immediately.  They didn't get particularly hot this time, but were definitely warmer than I'd expect them to be, given that they should be idle/disabled at startup.

And this time, to rule out anything stupid, I kept myself grounded with an antistatic wristband while setting-up, and I tested the new parts with just the bare minimum to see if the thing works:

* 24v power
* the 12864 display
* a few driver modules.

Nothing else has been connected so far -- no bed, no hotend, no motors, no end stops, no thermistors, no WiFi, no nothing, and I am certain the board's set up right -- see for yourself:

[image]
<sup>(there would normally be four drivers, but I removed one so that you can see the jumpers under it)</sup>

One bad motherboard or one dead driver I could believe, sure, but not this.  This would be two bad motherboards and 9 fried drivers, though since I never connected a motor to the new board, let alone tried to move any axis, hopefully none of the new parts are *actually* damaged yet.

I see only three possible answers:

A. There could be something wrong with my configs or build environment.
B. There could be a bug in Marlin, but Marlin devs are certain this isn't the case.
C. There could be a design flaw in the SKR v2 Rev. B that a few of us have been "lucky" enough to run into.

I just don't know what to do here.  I spent good money and wasted a lot of time on this, I just want it to work.

VanessaE, 2021-09-08:
[Marlin-configs-2021-09-08.zip](https://github.com/bigtreetech/SKR-2/files/7130460/Marlin-configs-2021-09-08.zip)

VanessaE, 2021-09-08:
As of right now, I have EIGHT dead drivers!

**THIS STUPID BOARD KILLS 2208's ON POWER-UP!!**

I cannot in good conscience recommend SKR v2 to anyone.

VanessaE, 2021-09-08:
@bigtreetech ...  guys, when you abandoned the SKR v1.3/1.4 design, you screwed up big time.

fitch22, 2021-10-04:
Hi All,
I have reviewed all these and related threads regarding the death of TMC stepper drivers.  I have been working on getting the TMC2209s to work with grblHAL, so its not Marlin but similar in many ways.  While working on this, I have managed to kill 9 TMC2209 drivers.  I really thought my software was somehow to blame.  I figured Marlin has working TMC2209 code, so I basically ported the software serial from Marlin to grblHAL.  Unfortunately, I killed another 3 TMC2209s.  It was time to look further.

grblHAL does not do a reverse stepper test.  But debugging my code I do restart the firmware often which toggles PC13 turning on and off the motor power.  This results in my board having the power turned on much more often than in a typical 3D printer booting up, where power goes on when Marlin boots, and things go off when the power switch is turned off.

I set up the board with code that just turns motor power on and off via PC13.  I monitored MGND as well as VBB, both with respect to GND.  I found that as VBB is ramping up, MGND has a nice spike of over 9V for many microseconds.  I believe this is more than sufficient to kill these stepper drivers.

Mind you, these scope shots were taken with no stepper drivers or stepper motors hooked up.  I would guess that due to trace lengths and PCB layout, that you could expect more or less of a spike at different stepper locations.  This may explain why some folks have only blown certain drivers and not all of them.  Possibly with stepper motors hooked up the spike may be different as well.  In my case, with no stepper motors connected, I had good luck with just 2 stepper drivers, but when I added the 3rd, they all blew.  I don't really want to blow up any more drivers so I am not keen on seeing what effect they have on the traces.  For me, seeing spikes like this tell me why mine blew.

I also jumpered MGND to GND and grabbed another scope shot.  This was part of the fix for rev A boards, but I think I would do this and disable the reverse test even with rev B boards.  I would suggest following BTT's guide for the rev A boards for jumpering.  I still saw 800mV with a single jumper, 2 could be better.  But for a permanent fix, I may simply pull Q1 and replace it with a fat piece of wire.

In both pictures, the green trace is PC13 turning on the motor power.  The yellow trace C1 is MGND referenced to GND, and the pinkish trace C3 is VBB also referenced to GND.

[image]

[image]

VanessaE, 2021-10-04:
Nice work finding the root of the problem!  Not happening the way I thought, but kinda close; that spike ion GND would definitely create a temporary path for a reverse voltage across the driver chip, with its various then-low digital IO lines being the other half of the circuit.

hcet14, 2022-02-20:
I just bought a SKR2. I guess you saved me a bunch of TMC2209s. Thank you all a lot.
Was there any suggestion from BTT how to fix this hardware bug? I know their suggestions for REV A, which are poor. 
Higher pricing boards like this one should be able to run all gimmicks from Marlin and TMC.

binhamed84, 2022-02-20:
Dear @hcet14 ,

About BTT SKR2 REV A:
They have three solution options
1. Replace
2. Fix
3. Return refund

For option "2" if you want to fix it, Check this link:
https://youtu.be/OH-h1Xz2V8g

Higher pricing boards: check Aliexpress most cheaper than other websites.
You will get the "board + tft35 E3 + TMC2209" for around 70$

hcet14, 2022-02-20:
Dear @binhamed84,

I won't start splitting hairs, but BTT states in the manual on page 2 "11) Introduces a new “anti-reversal” stepper driver protection which is achieved through a combination of new hardware and Marlin firmware. This protects against motherboard or driver damage due to incorrect driver insertion;"

A replacement doesn't fix this, the fix from BTT doesn't help to use DISABLE_DRIVER_SAFE_POWER_PROTECT. I don't think they will refund me since I bought a new, still sealed board with no original receipt and this is totally ok for me.

The price you stated above is to low. Cheapest I could find was about 93 Euros.

Maybe BTT will come along with a real fix.

neilmendoza, 2022-08-07:
Not specifically related to the SKR2 but I came a across this issue while searching for solutions to exploding drivers in my custom designed PCB. In the end, not setting the enable pin to low (enabled) before making sure that all the parameters had been set over UART seemed to fix my problem. Not sure if that's because I'm reducing the current over UART first or if it's just the extra pause before enabling that stops initial spikes hitting the motor coils and burning out the driver.

Mclute0, 2022-11-03:
> Nice work finding the root of the problem! Not happening the way I thought, but kinda close; that spike on MGND would definitely create a temporary path for a reverse voltage across the driver chip, with its various then-low digital IO lines being the other half of the circuit.

I had an extruder stepper driver die on a printer I bought new 8 days ago. (BTT BIQU B1 w/SRK2 Rev. b.) It died sometime after completing a five hour print. 

I inspected the driver and it looks like the TVS diode marked AE is cracked. I could replace it but I am old and suck at soldering SMD with pins and tiny zistors close by.

Or then again it is something else.

The heat sink was still really hot when I opened it up.

I opened a ticket with BTT, will see what they say.

[image]

stevewatts, 2024-06-09:
I found this post because I keep burning tms2208s too.  I have a standard ramps1.4 board.  
My question is what cooling do you have?

I went through a period of replacing x or y drivers every few prints, then ages with no problems.  Now I'm burning drivers and I think it's because the weather is warmer!

Stupidly my RAMPS board is upside down under the base plate with the thought that the head and heated bed would draw could air up through it.  This was probably a mistake.

So have a look at your cooling and see if opening your electronics casing and/or adding some fan cooling helps you