bigtreetech/SKR-2 #70: Endstops always triggered

phenix1990, 2021-09-15:
Hello all,

for a few days i try to fix my endstop problem. I have a CR10s and upgraded from the original mainboard to the skr2. I never had any endstop problems before.

My endstops are NC connected to Signal and ground of the endstop ports of the board. Pulldown is active.

Stange  thing is that when I connect signal direct to ground, M119 shows me contact open as expected. This should confirm the wiring is correct. So i measured the resistance of the endstops and got these values:
X endstop: ~20 Ω
Y endstop: ~  2 Ω

Are the values to high and therefore causing permanent triggered state? i thing 2 Ω should be ok.

Any ideas?

SMTDesigns, 2021-09-22:
just a thought... I am working on the marlin version for the SKR2. With an open connection and only ground to the endstop, you will need a pull-up on the pin in order to ever have a high level Also, the resistance measuring from a meter will apply a current to the circuit. You are not going to get a valid resistance to ground. You will see 0.2 ohms when you have the end stop switch "on". You need to measure voltage with a pull-up enabled. The voltage should be 3 volts or so when the switch is open and 0 when it is closed (with the board power turned on).

elvisior, 2021-09-24:
I think I have a very similar problem. 
I have a Delta that was previously using an skr 1.4 turbo (until I let the magic smoke out via a short from a faulty part cooling fan).
I can't get X and Y end stops to not always be in a triggered state. 
I was hoping to use optical endstops with this board but to remove variables I've put the jumpers on so I can use sensorless homing with my TMC 2209 drivers. 
Without any end stops connected it always shows X and Y to be triggered. And thus homing never works properly. This is the same behaviour I was seeing when I was trying optical and mechanical end stops.  Very frustrated. This might be the end of 3d printing for me. I've been through so many boards and printed so very little. :(

SMTDesigns, 2021-09-25:
It sounds like the mechanical endstops have not been disabled.
 
I tried the SKR2 but when the previously working thermistor would jump 10 degrees randomly when heating, I searched and found this was a error int he skr2 and there was a suggested soldering fix. I switched to the MKS Robin Nano V3.0 and haven't had any issues with anything yet. I am waiting for a new hotend to test everything.
 
-----Original Message-----
From: bigtreetech/SKR-2 
Sent: Sep 24, 2021 5:10 AM
To: bigtreetech/SKR-2 
Cc: Steve Thatcher , Comment 
Subject: Re: [bigtreetech/SKR-2] Endstops always triggered (#70)

 
I think I have a very similar problem.
I have a Delta that was previously using an skr 1.4 turbo (until I let the magic smoke out via a short from a faulty part cooling fan).
I can't get X and Y end stops to not always be in a triggered state.
I was hoping to use optical endstops with this board but to remove variables I've put the jumpers on so I can use sensorless homing with my TMC 2209 drivers.
Without any end stops connected it always shows X and Y to be triggered. And thus homing never works properly. This is the same behaviour I was seeing when I was trying optical and mechanical end stops. Very frustrated. This might be the end of 3d printing for me. I've been through so many boards and printed so very little. :(

elvisior, 2021-09-26:
If I try and disable endstops in marlin I get compilation errors that Delta's need endstops...

  In Configuration.h I comment out the following lines:
      
      //#define USE_XMIN_PLUG
      //#define USE_YMIN_PLUG
      //#define USE_ZMIN_PLUG // a Z probe
      //#define USE_XMAX_PLUG
      //#define USE_YMAX_PLUG
      //#define USE_ZMAX_PLUG

  Which results in the following compilation errors:

      Marlin\src\feature\../inc/SanityCheck.h:1449:6: error: #error "You probably want to use Max Endstops for DELTA!"

Or is there another way I should be disabling them?

elvisior, 2021-09-26:
I've tried the supplied marlin build in this repo and the endstops (manual or sensorless) don't work on it either for me.  Do all of the endstops x/y/z work for anyone?  i.e. is my board faulty or is the board flawed?

SMTDesigns, 2021-09-28:
personally, I prefer using electrical endstops rather than having a stepper controller "hit" something to figure out it is at an endstop location.  The delta homing switches are all at the MAX locations in the deltas I have used. Maybe you should try the electrical endstops and then move to "stress" endstops once you get the first ones working. Just a thought...

When I was working with the SKR2, I did get the electrical endstops working, so not sure what issues are going on for you.

phenix1990, 2021-10-01:
> just a thought... I am working on the marlin version for the SKR2. With an open connection and only ground to the endstop, you will need a pull-up on the pin in order to ever have a high level Also, [...]

Thanks for your response. I unplugged the endstops to measure the resistance of cables plus switch because I assumed a cable problem. Indeed x had a problem at a soldered extension.

The Y endstop issue I still don't understand. It worked after a firmware flash without any endstop related changes it stayed triggered. After another firmware flash without endstop changes it worked again. Really strange

So now it's finally working again without any problems

phenix1990, 2021-10-01:
> I've tried the supplied marlin build in this repo and the endstops (manual or sensorless) don't work on it either for me.  Do all of the endstops x/y/z work for anyone?  i.e. is my board faulty or is the board flawed?

Yes it can work. If you can't make it working with the endstops and still want to go with sensorless homing are you sure you activated sensorless homing correct? Maybe something is missing here to activate in the firmware config. When the error occurs that you need endstops you can click on the error Message which will navigate to where the error is thrown. For me it helped to check the transition for several errors to understand what other setting is related to this.

Also maybe you really need to enable the max endstops with the correct setting's.

Also maybe another Tipp: you find a lot of example firmware settings from Marlin, maybe you can compare some endstop related settings
https://github.com/MarlinFirmware/Configurations/tree/release-2.0.9.1/config/examples

I also watch a lot of example Videos for several hardware parts and did a lot of research on some problems, I hope you can get your printer run again

elvisior, 2021-10-02:
Is it possible I shorted my board?  I checked the wiring and I'd connected the power to the heated bed connectors by mistake (I feel like such an idiot...).  Could that explain the weird behaviour?

SMTDesigns, 2021-10-02:
If the board was truly shorted out, then the display would not work, etc.  Attaching power to the hotbed output really shouldn't do any damage. The - side is ground and the + side goes to just the mosfet that turns power on and off to the output. There isn't a path to the processor section from there. 
 
Have you tried just the mechanical end stops?  Usually, the end stop switch is off until the switch is activated by the stop being "triggered".  The switch closes and connects the line to ground. That is why it is important to have the end stop pullup be turned on and also to not have the pulldown turned on for the pin.
 
-----Original Message-----
From: bigtreetech/SKR-2 
Sent: Oct 2, 2021 6:25 AM
To: bigtreetech/SKR-2 
Cc: Steve Thatcher , Comment 
Subject: Re: [bigtreetech/SKR-2] Endstops always triggered (#70)

 
Is it possible I shorted my board? I checked the wiring and I'd connected the power to the heated bed connectors by mistake (I feel like such an idiot...). Could that explain the weird behaviour?

elvisior, 2021-10-03:
There must have been a path from the bed power connections as that is how board had been getting power all this time.

I've tried mechanical endstops, opticial endstops, connecting the pins to ground directly.
I haven't tried removing all of the stepper drivers and then trying them, I'll try that later today.
I wish there was a control board made for idiots like me that find a way to short everything.  I've got so many broken boards.

SMTDesigns, 2021-10-03:
I checked the schematic again. The "engineers" that designed it broke the whole thing up in sixteen sections on a single page... The + side of the HBED is a connection back to power in through a 15 amp fuse. Your ground is a bit more mysterious. The other HBED connection is to a MOSFET that goes to ground for power to be applied to the heat bed. Not sure how the board powered up with just two wires coming from the power supply to the two heat bed terminals.
 
First off, there are three connections for each endstop  top is  +5  ground is middle and bottom is signal.  If you put a meter on ground of the power supply. You should measure +5, 0 , and then a voltage 2.5 to 5 if the signal line is pulled up. If you measure +5, 0, and 0 it means that the line is either pulled down or floating.
 
Here is a link to the github board layout https://github.com/bigtreetech/SKR-2/blob/master/Hardware/BIGTREETECH%20SKR%202-Pin.pdf
 
-----Original Message-----
From: bigtreetech/SKR-2 
Sent: Oct 2, 2021 6:29 PM
To: bigtreetech/SKR-2 
Cc: Steve Thatcher , Comment 
Subject: Re: [bigtreetech/SKR-2] Endstops always triggered (#70)

 
There must have been a path from the bed power connections as that is how board had been getting power all this time.
I've tried mechanical endstops, opticial endstops, connecting the pins to ground directly.
I haven't tried removing all of the stepper drivers and then trying them, I'll try that later today.
I wish there was a control board made for idiots like me that find a way to short everything. I've got so many broken boards.

AnEn030, 2022-02-01:
i noticed that the discription on the pdf /Hardware/BIGTREETECH SKR 2-Pin.pdf are wrong. and they are wrong with my board and measurements.
The Endstops it is written das the endstop are in following order from top to bot  +5v ; gnd; signal. however on mine as well as the pictograme next to it the 5v+ and gnd are exchanged. thus gnd; 5+; signal... . can someone confirm this . I have an STM32f429 vgt6

Headrush69, 2022-02-02:
@AnEn030 I have 2 of the same board and the end stop pins match the pdf. From top to bottom: +5, GND, Signal

AnEn030, 2022-02-02:
The BIGTREETECH SKR 2-Pin.pdf is wrong because the square symbones indicate a ground compare it with the eg fan. So eather the textvit wrong or the pictograma is. The orientation I am used from the skr 1.4 is top to bottom  (5v/gnd/signal)... Anyhow I measured the voltage and at least for mine it is wrong.  I'll post an picture later

Headrush69, 2022-02-02:
Yes, if the square indicates GND, then the pictogram is wrong, but I'm not sure that it does and that it's not just an orientation symbol for the connector.

AnEn030, 2022-02-02:
I just want to get the knowledge that my SKR2 Board does NOT provide the +5V on the top position. This knowledge took 3 hours to find and verify.  And i hope that this post will help someone!

Gipielle, 2022-02-27:
> Thanks for your response. I unplugged the endstops to measure the resistance of cables plus switch because I assumed a cable problem. Indeed x had a problem at a soldered extension.
>
> The Y endstop [...]

Hello, can you please share your CR10s config files so i can compare with mine? I have problem with z axis (i used a mix of skr 1.4 turbo)
Thank you