bigtreetech/SKR-2 #71: ESP 8266 ESP-07s

qulet, 2021-09-15:
The core of the ESP8266 ESP-07S module in the slot BTT SKR2 VB boards - the printer is not recognized in the WIFI network.
What should i do? Please provide information step by step to change this.
Marlin 3d v2.9.01 configuration files:
[MARLIN_config.zip](https://github.com/bigtreetech/SKR-2/files/7172734/MARLIN_config.zip)

qulet, 2021-09-17:
Please provide the instructions. How to program the WIF ESP07S module. What is the setting of the jumpers on the SKR 2 board for the above-mentioned module?
What needs to be changed in the Marlin 3D configuration code?

qulet, 2021-09-17:
Where should I look for this information?

LawnMo, 2021-09-18:
the wifi module is mostly for RRF usage, if you want to use it on marlin you need to breadboard it and flash ESP3D on it.  
Under marlin, wifi/U-disk jumpers don't matter, put them in U-disk as you won't be able to use SPI to connect to the ESP module (only UART will work).  
Once your ESP module runs ESP3D, you can plug it back on the motherboard and marlin only needs one change, you need to define a serial port to "3". (either SERIAL_PORT_2 or _3).  

Apart from monitoring the printer, don't expect much from it, upload speeds are as fast as your UART connection to it, it's just slow and not much can be done about it.  

Have fun, hope this helped :)

qulet, 2021-09-18:
Dziękuje za przekazanie informacji !  Moje pytanie  dotyczy fleshowania  ESP3D z poziomu VISUAL STUDIO CODE.  Jak w Marlinie dodam SERIAL_PORT_2 to po kablu USB  i włożonym  module ESP8266 ESP-07S w slot płyty SKR2, będę mógł wgrać firmware ESP3D ?

Matze1811, 2021-11-14:
Hello, I have the same problem. I bought an original BIGTREETECH ESP8266, but it doesn't work. Does anything still have to be flashed or configured here? Where can I find documentation? Regards

LawnMo, 2021-11-15:
> Hello, I have the same problem. I bought an original BIGTREETECH ESP8266, but it doesn't work. Does anything still have to be flashed or configured here? Where can I find documentation? Regards

The module comes preloaded with the wifi firmware for RRF, you need to flash esp3D on it, and you need build marlin with a serial port defined to 3 (to connect to the wifi module).  
Google "esp12 breadboard flashing", you can't flash a firmware to it from marlin afaik. You'll need some extra hardware to flash it, sadly there's no "specific documentations" for this, BTT being BTT, they mostly just release products and expect the community to do their work.

Gipielle, 2022-04-14:
same problem here