bigtreetech/SKR-2 #75: Low voltage on tmc2208 potensiometer

Dankoman, 2021-09-22:
I am in the process of replacing my SKR 1.3 with a SKR 2. Even though I bought the card a few weeks ago I got a rev A board.
Anyway i did the jumper wire grounding so it should be usable. Flashed the latest Klipper 3D.
I conected up 4 tmc2208 drivers to my delta and started to calibrate. everything seemed OK.
Then i started a test print and got error 
`firmware: TMC ‘stepper_b’ reports error: GSTAT: 00000001 reset=1(Reset) `
after a few seconds of printing.
Checked all my cabeling and even replaced all the stepermotor cables but the error persisted. Calibration movements work fine, manual movments from octopi works fine, printing dies after 5-10 seconds. Allways stepper_b...
Went on to check vith my voltmeter whar the potentiometer on the drivers was set to. 0.2V on all of them, tried to crank that up to the recomended 0.8-ish but it won't go higher than 0.36 then it rolls over to zero.
So instead of 0.0 - 2.0V mine goes from 0.0 - 0.36V.
Any ideas?

anarch79, 2022-03-31:
Check the cap between vm and ground

M3skar, 2022-08-30:
@Dankoman did you solve it ? i have same issue with an Anycubic Mega S and tmc2208, one the drives stopped going above  0.4V starting with the second measurement ( first measurement was good )