bigtreetech/SKR-2 #79: BTT SKR 2.0 Rev.B + TMC2209 on Anet A8Plus  Sensorless homing issue Marlin Bugfix 2.0

SergedeParis, 2021-10-19:
I have setup my SKR 2.0 Rev.B on Anet A8 Plus and everything seems to work fine. Setup includes 3D-touch and double Z stepper drivers. But when I try to use the sensorless homing after installing the mini jumpers on X and Y, I cannot get homing to work. The stall sensitivity seems to do something but is not reliable. It goes to No Stop (sensitivity at 103) to No Move (at 104). After restart the same Stall sensitivity changes from No Move to No Stop without changing anything.
I do not know what I am doing wrong.  Is there any other wiring change than the mini jumpers to have it to work? Could it be wrong TMC 2209? Help is welcome. 
I attached (and renamed them as .txt)  the Configuration.h and Configuration_adv,h files.

[Configuration.txt](https://github.com/bigtreetech/SKR-2/files/7375373/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/SKR-2/files/7375375/Configuration_adv.txt)

SergedeParis, 2021-11-24:
After checking other printers Marlin setup, it seems I have to work around TMCs current,  X at 500mA and Y at 600mA seems to give some good results, Then threshold at 80 on both axis look good.

roelfsre, 2022-02-26:
What happens when you disable IMPROVE_HOMING_RELIABILITY ? (configuration_adv.h)
I just rebuild my Anet A8 Plus with the SKR2.0 board Rev B and still tweaking, but had some issues homing when I had IMPROVE_HOMING_RELIABILITY enabled. Might give it a try.
Sensorless homing does work now, I have TMC 2209. Sensitivity set to 145 for both axis. TMC current left on 800mA.

roelfsre, 2025-12-29:
This is Marling 2.1.2.6 config

[Configuration.txt](https://github.com/user-attachments/files/24374713/Configuration.txt)
[Configuration_adv.txt](https://github.com/user-attachments/files/24374712/Configuration_adv.txt)