bigtreetech/SKR-2 #82: SKR V2.0 and STM32f429

Greyukr, 2021-10-31:
Ask for help!
My English is so...
A few days ago I received a SKR V2.0 with STM32f429. This is a new chip. Marlin doesn’t have that chip yet. 
I’m compiling a firmware based on F407. Right?
I want to use BLTouch without using the Z-axis concept contacts.
What should I write here:
#define Z_MIN_PROBE_PIN - ?????????????????

Best regards,
Sergio!

karabas2011, 2021-11-08:
Please post a link to this board.
I see only STM32F407  on SKR 2

karabas2011, 2021-11-08:
Technical specifications
In the following list you can find the technical specifications of the **SKR 2** electronics:

Size 110mmx85mm, in the previous section you have a diagram of the board
ARM Cortex-M4 **STM32F407VGT6** MCU at 168Mhz

Greyukr, 2021-11-08:
https://www.biqu.equipment/products/bigtreetech-skr-2?_pos=1&_psq=skr+2&_ss=e&_v=1.0&variant=39444693188706
Please note that the F407 chip has been upgraded to the F429 chip. If you
order the motherboard, the F429 chip will be sent by default. The two
control boards have the same functions except for the different chips.
I bought here and I got 429. There’s a note below.
https://aliexpress.ru/item/1005002483248701.html?spm=a2g39.orderlist.0.0.3fee4aa6s7ykUt&_ga=2.1390453.53312825.1636289231-1470277378.1629784519

karabas2011, 2021-11-12:
Thank you, now I see.
As I see there are still no platformio environment for this processor.
And it is not in list of supported ones. We need wait for someone who will build them

Greyukr, 2021-11-12:
I’m sorry I didn’t buy it - Makerbase MKS Eagle-
https://www.aliexpress.com/item/1005003352842756.html?dp=5f3ad6e098283b11d559b783&cn=ah&aff_fcid=652ee38c1fe34a1f851dc923b8eb4e89-1636735970165-07585-_AnTGXs&tt=CPS_NORMAL&aff_fsk=_AnTGXs&aff_platform=portals-tool&sk=_AnTGXs&aff_trace_key=652ee38c1fe34a1f851dc923b8eb4e89-1636735970165-07585-_AnTGXs&terminal_id=d2fb5bd3840645278757b4e969f71614

radek8, 2021-11-16:
> Ask for help! My English is so... A few days ago I received a SKR V2.0 with STM32f429. This is a new chip. Marlin doesn’t have that chip yet. I’m compiling a firmware based on F407. Right? I want to [...]

[image]

marcossfernandez, 2021-11-18:
Same here with SKR2 Rev.B with a STM32F**429** board.
I don't know what modifications need to do to the platform.io to previous STM32F407 chip.

We are suffering a lot of USB disconnects from Raspberry's Repetier Server. May it could be for using STM32F407 old chip for firmware building?

Klemens75, 2021-11-18:
I have the same problem with F407 to F429. By Marlin´s board.h - list  is SKR-2 Rev.2 with STM32F407 chip. But i needet with STM32F429VGT6. 
Can anyone help me there. For that I have too little knowledge
[image]
.

Klemens75, 2021-11-18:
[image]

Klemens75, 2021-11-18:
There are still no platformio environment for this processor. And it is not in list of supported ones. 
Who could help with that?

marcossfernandez, 2021-11-18:
I could compile it and make it work with the SKR 2 (407) current processor in platform.io. Aparently all functions are ok, but i don't know if lack some other things or if performace is the right one. 

It moves, heat and does the basic things, but it isn't ideal, of course. The new SOC has some instructions and memory who it can't be accessed. I've asked Biqu for help but no sucess at the moment.

Klemens75, 2021-11-18:
For me, the difference is too great. The F429 could do much more. 
[image]
[image]
In detail it is much more.

marcossfernandez, 2021-11-19:
Biqu guys had sent this link to me after contact them yesterday. Awesome response. It comes with a Marlin last version with F429 enviroment:

https://github.com/bigtreetech/marlin/tree/B1-2.0.9.2.x

You ought to modify and suit your needs.

Greyukr, 2021-11-20:
> Biqu guys should have made a Rev. C

marcossfernandez, 2021-11-22:
https://github.com/bigtreetech/SKR-2/tree/master/Firmware

> [platformio]
> src_dir      = Marlin
> boards_dir   = buildroot/share/PlatformIO/boards
> default_envs = BIGTREE_SKR_2_F429
> include_dir  = Marlin

This version its the latest bugfix, and corrected F429 enviroment. This one works as expected with the new SOC version.

JuanPuchol, 2021-11-22:
Good news,

the code has been updated and now envireonment BIGTREE_SKR_2_F429 can be used!!!

jamesjnadeau, 2021-11-25:
Is there a pull request against marlin I can follow for this?

alienboyxp, 2021-11-25:
> Is there a pull request against marlin I can follow for this?

https://github.com/MarlinFirmware/Marlin/pull/23177 @jamesjnadeau  I think this is the referred to the issue

Greyukr, 2021-11-26:
Over a maximum of five minutes, several random reboots of this motherboard takes place. What is the cause of these reboots?

sbaeder, 2021-12-08:
On the BLTOUCH - I had issue using the "default" plug after it was working before using Z_MIN...In the firmware on the BTT github...It is mapped in the pins.h file for the board.  BUT, I switched back to the z-min plug, and defined it in the configuration.h file
`#define Z_MIN_PROBE_PIN Z_MIN_PIN // Pin 32 is the RAMPS default`
But if ou look at the pins files, they are basically the same - most in a common file...

sbaeder, 2021-12-08:
> > Biqu guys should have made a Rev. C

Technically, the BOARD is the same.  Chip pin out is the same, so no real need to revise the environment...

Use the Marlin AUTOBUILD stuff.  Makes it simple to invoke the build, and the board and type is listed fine in the code from the Bigtree github

sbaeder, 2021-12-16:
> I installed the latest Marlin software for this board. The board reboots every three minutes. I disabled the drivers, WIFi, TFT, but it didn’t work. I need your help! What could be the reason?

As mentioned above, we had another issue on the reboots.  Looks like removing the SDcard will prevent it from happening.  Possible that there is an issue in the firmware (Marlin) after flashing/booting that is causing it.  Until there is more "proof" (or a better fix), just pull the card after flashing...

Greyukr, 2021-12-16:
The reason for the reboot  were: 
Configuration_adv.h
#define USB_FLASH_DRIVE_SUPPORT
#define USE_OTG_USB_HOST
#define NO_SD_DETECT
#define MULTI_VOLUME
I closed those lines and the reboots stopped

bfmvtony, 2021-12-18:
> I installed the latest Marlin software for this board. The board reboots every three minutes. I disabled the drivers, WIFi, TFT, but it didn’t work. I need your help! What could be the reason?

So this finally works? what things they changed? Do you need to comment the lines you wrote in your last post? 

Thanks in advance!

sbaeder, 2021-12-19:
I have continued to work this issue, and I *THINK* it might be a conflict between the onboard USB trying to be used vs. the onboard SD card.  I think that if BOTH are enabled, it is a conflict on the software that may cause this.

If I try to have the above "OTG" things as well as the  "ONBOARD" value for the SD Card, it will cause reboots.  A switch to the LCD value "seems" to fix this.  I also see in a diff of the original BTT files from their github that a default is LCD, but in the version committed to marlin, that it is ONBOARD...

Still need to look at this more.......

Cabinfourus, 2021-12-24:
I have one of the SKR V2.0 with STM32f407 boards. I have been setting up Marlin and have yet to power the board on. Before I do anything else I'd like to know if there is anything I need to change in Platform I/O. 
This is what I have it set to:
#  default_envs = BIGTREE_SKR_2
#  default_envs = BIGTREE_SKR_2_F407
Are there any other changes I need for this board to work correctly?
Does the SKR_2_F429 environment work with the 407?
My printer setup is: 
All BTT 
SKR 2 (STM32F407) FET HY1904 
UPS power fail unit
Relay V1.2
TMC2209 using sensorless homing
Filament sensor (not BTT)
3D Touch used for homing to the middle of bed and auto-leveling
Marlin version is from the attached files in SKR 2 from GitHub

bfmvtony, 2021-12-24:
Check serial port.. with skr cannot be 0. Imita be from 1 to 6 or -1

bfmvtony, 2021-12-24:
> I have one of the SKR V2.0 with STM32f407 boards. I have been setting up Marlin and have yet to power the board on. Before I do anything else I'd like to know if there is anything I need to change in Platform I/O. 
> 
> This is what I have it set to:
> 
> #  default_envs = BIGTREE_SKR_2
> 
> #  default_envs = BIGTREE_SKR_2_F407
> 
> Are there any other changes I need for this board to work correctly?
> 
> Does the SKR_2_F429 environment work with the 407?
> 
> My printer setup is: 
> 
> All BTT 
> 
> SKR 2 (STM32F407) FET HY1904 
> 
> UPS power fail unit
> 
> Relay V1.2
> 
> TMC2209 using sensorless homing
> 
> Filament sensor (not BTT)
> 
> 3D Touch used for homing to the middle of bed and auto-leveling
> 
> Marlin version is from the attached files in SKR 2 from GitHub
> 
> ** When I compile Marlin using #  default_envs = BIGTREE_SKR_2_F407 in the ENV, it does succeed. 
 
--- Check This ---
Check serial port.. with skr cannot be 0. Imita be from 1 to 6 or -1

sbaeder, 2021-12-25:
> Does the SKR_2_F429 environment work with the 407?

no - there are two different "environments" - one for either chip.

bfmvtony, 2021-12-25:
> > Does the SKR_2_F429 environment work with the 407?
> 
> 
> 
> no - there are two different "environments" - one for either chip.

Well, I tested it and works, but not correctly. As someone said before, it suddenly reboots. So better use the right environment.

buildjournal, 2022-01-06:
Okay I got the new board with F429. How do I download this folder: https://github.com/bigtreetech/SKR-2/tree/master/Firmware/Marlin-bugfix-2.0.9.2.x

Greyukr, 2022-01-06:
[image]
[image]

Greyukr, 2022-01-06:
[image]
[image]

buildjournal, 2022-01-06:
I did that and I got the files.
Still getting errors compiling.
Do you have an Ender 5 Plus specific configuration?

sbaeder, 2022-01-07:
How are you building it - are you using the Marlin auto build?  If so, be sure to select the F429 based onfiguration.  Also, note that there is an "official" Marlin version to support this alternate chip that was released by Marlin on Christmas day (2.0.9.3)

buildjournal, 2022-01-07:
Yeah I did all that. Using your Marlin '2.0.9.2.x & F429' folder from
Github and set the environment to 'default_envs = BIGTREE_SKR_2_F429'.

I guess what I'm saying is, I can successfully build a stock Ender 5 Plus
firmware with no modifications at all. There are sample configuration.h and
adv.h on Marlin Github.

However I am trying to replicate the firmware settings from my F407
firmware. I got tons of community help to build the F407, but the F429 is
so new I don't know how to compile. There is no firmware support for
dual-z, etc because it is so new.

jamesjnadeau, 2022-01-07:
I've been able to build this successfully for the F429 chip using dual z axis motors. Runs great. So it's something in your config that is causing it not to build. I was able to get it to build using the latest marlin bugfix branch and the code available in this repo(both with my custom machine config)

You're settings are either old, or your misconfiguring something this board requires. Read the terminal output of the build, look from some red and yellow lines toward the end. I've found the error message from the build process to be really helpful in determining the problem I'm encountering. 

I started with ender 3 config. Please note that some variables have changed names since the original 2.0 release your ender firmware is likely based on. There's also a config variable that has a date stamp in it that will fail the build if it doesn't match what the codebase is expecting.

Read the compile log output, it's there to help

sbaeder, 2022-01-07:
or possibly post the log for someone to look at it an help.  If you were able to do it for the older chip, AND it's using supported code (as opposed to a modified set of Marlin code), then it should compile.

buildjournal, 2022-01-07:
Can you send me your folder to compile? Including platformio, config.h,
config.adv.h? I understand your build is different but maybe I can start
there.

jamesjnadeau, 2022-01-09:
@buildjournal Here's my configs:

https://github.com/jamesjnadeau/Marlin_configs/tree/master/SKR-2

I only changed this line in platformio:
```
default_envs = BIGTREE_SKR_2_F429
```

buildjournal, 2022-01-11:
Thank you very much. Hope this works :)

SpectralCreations, 2023-03-02:
i too have the SKR2 revb and im having trouble getting this to compile 
https://github.com/SpectralCreations/SKR_2_F429
i keep getting "Error: Could not find one of 'package.json' manifest files in the package" 
and i have no idea what im doing