bigtreetech/SKR-2 #86: BTT SKR2  VB ERROR

qulet, 2021-11-14:
Printer stopped printing - extruder warming up error. The LCD screen keeps blinking like in the video in the attached link:

https://drive.google.com/file/d/1QjQQAvrOKwgDH_WacmhPQEEnJOQDrkMy/view?usp=sharing

Replacing the LCD, removing stepsticks, or reloading the firmware does not remove the problem.

radek8, 2021-11-16:
I'd guess it's short on the encoder.
You say that replacing the LCD didn't help.
Then there could be a short circuit on the ribbon cable to the LCD.
Try replacing that cable
If it is not a cable, then there could be a short circuit on the SKR board.
I would try to remap the pins to the encoder in FW. If it stops, it would be the solution ....

qulet, 2021-11-17:
Thanks for your support and answer
I checked the tapes connecting the LCD with the BTT SKR2 are OK. How do I do this pin remapping?

qulet, 2021-11-17:
*               -----                                             -----
 *                NC | 1 2 | GND                                                     5V | 1 2 | GND
 *           RESET | 3 4 | PC4 (SD_DETECT)              (LCD_D7) PE13 | 3 4 | PE12 (LCD_D6)
 *   (MOSI) PA7 | 5 6    PB2 (BTN_EN2)                (LCD_D5) PE11 | 5 6   PE10 (LCD_D4)
 *  (SD_SS) PA4 | 7 8 | PE7 (BTN_EN1)                   (LCD_RS)  PE9 | 7 8 | PB1  (LCD_EN)
 *     (SCK) PA5 | 9 10| PA6 (MISO)                    (BTN_ENC)  PB0 | 9 10| PC5  (BEEPER)
 *               -----                                             -----
 *               EXP2                                              EX
[pins_BTT_SKR_V2_0_common.txt](https://github.com/bigtreetech/SKR-2/files/7555095/pins_BTT_SKR_V2_0_common.txt)
This is my PINS file, I understand that you need to replace BTN_EN1, BTN_EN2, BTN_ENC.
I don't know for what?

radek8, 2021-11-17:
Which connectors do you not use on your SKR2 and could they be used for remapping?

radek8, 2021-11-17:
Send a photo of the connected board, I will try to find a free pin according to the photo.

qulet, 2021-11-20:
I changed PIN_y:
  PE7 to PE15
PB2 to PB14
PB0 to PB10 (WIFI PINS)
This did not fix the problem.
As part of the complaint, the album was sent back to the seller.

closed 2021-11-20 by qulet