bigtreetech/SKR-2 #89: The Marlin Firmware doesn't flash with sd card

Daniele981, 2021-12-02:
Hi,
i have a problem on my skr 2 STM32F429VGT6 to flash marlin with the sd card.
The firmware is the lastest bugfix with BIGTREE_SKR_2_F249 platform.io configuration.
I formatted the SD card in fat32 , loaded the "firmware.bin" file and i put in but nothing when reboot the board doesn't flash it!
the SKR is powered by the 24 v power supply.

Can you give me some help?

There is another way to load firmware without sd card?

Motobluesman, 2021-12-02:
I have the same problem with a new board.. with TMC2208..

Daniele981, 2021-12-02:
I think that the problem is about the marlin firmware 2.0.9.2...
Now I try with an old bugfix

radek8, 2021-12-03:
How big is your SD card? Use 8GB or less!

Daniele981, 2021-12-03:
The SD is 8 GB!

sbaeder, 2021-12-08:
Try hitting the reset button on the LCD screen - mine also seemed to not flash the first time until I did that.

Donpilkerton, 2021-12-13:
Same issues. I can not get the skr 2 or accept a marlin sd flash or a klipper firmware.bin flash. Ender 3 pro stock screen I’ve used in my skr 1.3 for years.

Motobluesman, 2021-12-13:
resolved
the problem was in the micro sd..
when you formatting is very important,,first... fat 32 and second... 
unit allocation 4096 byte

Donpilkerton, 2021-12-13:
Motobluseman that is 100% incorrect on the issue being fixed. I have formatted my sd card correctly prior to you suggesting that and it still will not pull the update off the sd card. I have a ton of people telling me that there is a known issue with these boards and the sd card slot.

Daniele981, 2021-12-13:
Yep Donpilkerton is right!
Yes, I was able to ascertain that the problem is due to the SD slot...

Donpilkerton, 2021-12-13:
Daniele981 were you able to get into contact with anyone about a replacement. I guess the one good thing about it is I bought it from the BTT store on amazon so I can probably just return it for a refund if I am not able to get an update on it.

Daniele981, 2021-12-13:
I will try!

closed 2022-01-23 by Daniele981

mayder, 2022-09-21:
I have the same problem, my SKR Mini E3 V3 don't start with SD Card and don't read firmware.bin file. I tried send using vscode with "autobuild marlin" plugin but I didn't succes. Today I bought stlink hardware for try update my firmware.

Donpilkerton, 2022-09-21:
> I have the same problem, my SKR Mini E3 V3 don't start with SD Card and don't read firmware.bin file. I tried send using vscode with "autobuild marlin" plugin but I didn't succes. Today I bought stlink hardware for try update my firmware.

Try using a 1gb sd card or smaller. That is what I had to do to get the flash to work properly. It would not complete with anything over a 2gb card for some reason. All of my problems are were with an skr2 but it may work for you.

mayder, 2022-09-22:
Thanks, I will try

mayder, 2022-09-22:
My sd card have 8gb, I will try with a smaller

mayder, 2022-09-22:
> @mayder Try using a 1gb sd card or smaller. That is what I had to do to get the flash to work properly. It would not complete with anything over a 2gb card for some reason. All of my problems are were with an skr2 but it may work for you.

I tried with SD Card 256mb, but I always receive message "SD Init Fail".
I formated SD Card with Fat, Fat32, NTFS, exFAT, but all fail.

rio1711, 2022-09-23:
> I have the same problem, my SKR Mini E3 V3 don't start with SD Card and don't read firmware.bin file. I tried send using vscode with "autobuild marlin" plugin but I didn't succes. Today I bought stlink hardware for try update my firmware.

I have the same problem with my ender3 v2  SKR Mini E3 V3 don't start with SD Card and don't read firmware.bin file.
and when i print with sd card the printer reboot randomly

schrammi1996, 2022-11-15:
i had the same Problem with an btt skr 2 and the solution was just to delet the .bin after firmware. 
Just name it firmware

sbaeder, 2022-11-15:
as I understand it, a lot of the issues are related to how it reads files on the sdcard, so it's not a surprise to me that this fixes it.  I saw (but can't seem to find it now) a set of posts in the Marlin discord or github about some timing issues in accessing the SDCARD and adding things like the checksum feature and a few other changes to the configs can help here as well.

I'll try more to find that and link this to those posts...but making changes to the files on the sdcard as well as how Marlin interacts with the sdcard or just pulling the sdcard all seem to "fix" the issue

wambs8, 2023-04-22:
> i had the same Problem with an btt skr 2 and the solution was just to delet the .bin after firmware. Just name it firmware

That did it for me. Everyone says to do Firmware.bin but that doesn't work

hariprasadP3D, 2025-09-14:
> resolved the problem was in the micro sd.. when you formatting is very important,,first... fat 32 and second... unit allocation 4096 byte
This worked!!!!! , I thought it was a corruption in the firmware and was trying to re-flash it using an ST link. Thank you so much!!!.