bigtreetech/SKR-2 #90: Z and X moving wrong direction

mechmann, 2021-12-06:
Purchased SKR2 and TFT screen.  Assembling to an Ender 5 Plus.  Using 2209 drivers.

Compiling Marlin from the bugfix version present on here.

X and Z are moving the wrong direction when I command them to move on the TFT.  I have tried inverting the axes and it has no effect.

Also, when I change to Marlin mode on the TFT I get nothing.  Is there another setting in Marlin that configures the coordinate system?

Please help.

iamthebest22, 2021-12-06:
Oh wow, hey I'm having a similar issue on the Z, I did invert the direction of my Z so normal movement is fine, but when I want to home Z, it moves in the wrong direction and then stalls and halts itself. Using SKR 2 with the 429 chip and TFT35 MZ V3.0 here. Someone could help us both here? Nightly build bug?

mechmann, 2021-12-06:
What did you do to invert your motion?  I have changed the setting (INVERT_Z_DIR) and it has no effect!

mechmann, 2021-12-06:
I am starting to think that even though the .bin file is gone from the MicroSD card the firmware isn't changing.

iamthebest22, 2021-12-06:
the invert_Z_DIR, changed did it for me, but when I try to use the TFT 35 MZ to probe z, it goes in the wrong direction. Wonder if my MZ screens needs to update firmware. Firmware is updating fo rme, but yeah, I'm getting stall and then halt when I try to probe, and it's not the endstop issue.

mechmann, 2021-12-06:
Did you change the probe direction?  Seems like you are closer to getting it correct than I am.   Invert your Z Probe direction (1 or -1).

iamthebest22, 2021-12-06:
that's what I didn't change because it says in the file, 1 is for max, -1 is for min for Z homing direction. Currently if it's at -1, and I change it to 1, won't that make it think it's going to max after homing?

mechmann, 2021-12-06:
if it is going the wrong way for homing i think that would be the one to change.

iamthebest22, 2021-12-06:
okay, but I still think I would get the stall and then halt issue, but it's late here, I'l try it tomorrow and report back.

mechmann, 2021-12-06:
I am fairly certain that my board is not flashing the new firmware even though the firmware.bin is gone and I have firmware.cur on the card.  I am hoping that BTT can help me debug this and fix it.  I am using the firmware from this github source.  I have tried compiling with zero changes and with the only change of inverting the axes.

radek8, 2021-12-06:
The direction of movement is set:
#define INVERT_X_DIR (true or false)
the location of the end stop is here:
#define X_HOME_DIR (-1 = min, 1 = max)

radek8, 2021-12-06:
You must first set the correct direction in the Marlin environment, then first set the correct direction on the TFT touch screen

radek8, 2021-12-06:
The Z axis moves in the correct direction on the touch screen.
The display is designed for a moving nozzle. When moving down, the nozzle moves towards the bed. You have an Ender 5 with a moving bed and it confuses you. Just rename the icons (swap names)
z_dec.bmp
z_inc.bmp

iamthebest22, 2021-12-07:
so I did the update, and change stuff, still no avail. I have it connected to Pronterface and also tried Mattercontrol. all movements are fine, but when I go to home just Z, it still moves int he wrong direction. 
can't change define Z home direction to 1 because then it won't compile and says that's going to max. I'm at a loss.

A note: I'm using Precision Piezo Orion, which uses the nozzle as the probe. nothing is triggered during the homing session, however, when the direction is going the wrong way during z homing, i triggered the Z piezo, but nothing happened and it kept going and I had to shut down power or emergency stop.

sbaeder, 2021-12-13:
Where/how is the Piezo plugged in and configured?  Z-Min?  Make sure that section of the config is mapping to the right place when configuring the "probe"

bfmvtony, 2021-12-18:
In config.ini from sd card firmware you can also change de direction of the buttons from 0 to 1 so it is easier than change pics I think, and it was useful for me.

tattoofreak098, 2021-12-19:
[Configuration.zip](https://github.com/bigtreetech/SKR-2/files/7741787/Configuration.zip)

ender 6 hatte auch das problem achsen hauem jetzt alle hin allerdings funktionieren die endstopps und homing nicht bei diesem board bekommt mein hund auch graue haare 

vielleicht hilft die config

radek8, 2021-12-20:
Write in English