bigtreetech/SKR-2 #91: SKR2 restarts

oleksey, 2021-12-12:
Hey, 

I've got SKR2 Rev B with TFT35 E3 V3.0 and TMC2208 drivers, flashed Marlin 2.0.9.1, but once turned on printer restarts every couple of minutes. I don't understand why and don't know where to start looking for problems.

- Drivers are in UART mode
- changing drivers to TMC2226 didn't help
- flashing marlin from bugfix branch didn't help
- flashing marlin from 2.0.8.x branch didn't help
- flashing marlin taken from a friend who has same setup, but working
- Checked the voltage that goes from PSU to SKR2, and it is 24.2V
- When SKR2 restarts there is no dip in voltage - it constantly remains 24.2V - so it doesn't look like PSU problem
- Octoprint doesn't tell anything except this:
`Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3940 Please see https://faq.octoprint.org/serialerror for possible reasons of this. Changing monitoring state from "Starting" to "Offline after error" Connection closed, closing down monitor`

Configuration.h - https://pastebin.com/0ZYPRjtk
Configuration_adv.h - https://pastebin.com/MtMD64c3

Here is an example how it all goes: https://youtu.be/2pnIN2fDaCo

Any ideas what else can I check/do? I just want it to work already... :D
BTW, the printer is Enter3 Pro.

sbaeder, 2021-12-13:
While I don't use the TFT (just an older 12864 display via exp1/2) I also am seeing the same sort of issue.   After a while, it seems to settle down and stop rebooting, and can run a print, but after a power off, it repeats the same issue of rebooting spontaneously.  See same errors using Pronterface.  Basically this is the message when it can't communicate due to reset.

oleksey, 2021-12-13:
> You didn't say, but is this with the 407 or the 429 chip. Both are "rev B" boards. Mine is the 429.

That is actually a good point. Mine is also 429.

I have a board with 407 chip laying around, and now that you are saying it could be something chip related, I should probably test the other board to see if there is any difference.

sbaeder, 2021-12-14:
Heard back from BTT support.  Suggestion to see if putting the jumper for board electronics "power" to the USB instead of the external 12/24v   If it is some sort of a "glitch" in the conversion to the 3.3v power for the processor, this might help ID the issue.

Going to try that later today

oleksey, 2021-12-14:
Something I have found out today [here](https://github.com/bigtreetech/SKR-2/issues/82#issuecomment-975519931):
bugfix version I've tested apparently wasn't latest and did not include F429 chip:
```
#define BOARD_BTT_SKR_V2_0_REV_A      4212  // BigTreeTech SKR v2.0 Rev A (STM32F407VGT6)
#define BOARD_BTT_SKR_V2_0_REV_B      4213  // BigTreeTech SKR v2.0 Rev B (STM32F407VGT6)
```

but the "latest latest" bugfix version now mentions F429:
```
#define BOARD_BTT_SKR_V2_0_REV_A      4212  // BigTreeTech SKR v2.0 Rev A (STM32F407VGT6)
#define BOARD_BTT_SKR_V2_0_REV_B      4213  // BigTreeTech SKR v2.0 Rev B (STM32F407VGT6/STM32F429VGT6)
```

Haven't tested that yet, but I hope that will sort things out.

oleksey, 2021-12-14:
latest Marlin build that has F429 did not help, printer still restarts.
The source code was taken from BTT repository.

I guess I have to test F407 out.

oleksey, 2021-12-15:
Apparently once you remove the SD card from the slot it stops resetting itself.

closed 2021-12-15 by oleksey

sbaeder, 2021-12-15:
Tried this on mine too, and it is OK as well.   Maybe issue in Marlin w.r.t. the handling of the sdcard?