bigtreetech/SKR-2 #92: SKR2 and TFT35 no printer attached!

Cr0nJ0b, 2021-12-16:
Hey Folks,

I've been struggling with no printer attached for my new SKR2 board.  I have setup everything correctly as far as I can tell.

What I've checked so far:
1) Cables are correct -- Even tried inverting the RS232 cable, no joy.  ext1 and ext2 are correct.  I've checked voltage, just to make sure it's all good.
2) In marlin I have the Serial_Port set to 1, SP2 set to -1 and SP3 set to 2 all with 115200 as the baud.  I've tried changing the baud on both the TFT and marlin, no joy.  I've tried changing the SP to various setups including 0, -1 and 1, -1 and -1, 0 and -1, 1.  According to BTT PDF for this combination (SKR2 and TFT35 E3, V3) #define SERIAL_PORT X should be 1 and the board type is #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER -- no joy
3) I tried a different SKR2 board with the same firmware -- no joy.
4) Marlin mode does function correctly, with minimal testing
5) I updated the firmware in for the TFT to the latest version with the correct settings for Printer baud etc.  I've also tried changing baud rate once the printer starts.
6) in all cases, I wait about 2 min with the no printer attached screen before moving on.  In all cases, I've tried USB power and 12V power.  using marlin 2.0.9.2

Things I've observed:
1) sometimes with various SP settings i've noticed that Marlin mode didn't work when cables were attached.  but marlin mode is working fine now.
2) the 5V pin always reads 1.3V for both SKR2 boards.  Not sure if this is something that is just normal, but both boards had the same behavior.
3) When I attached the new SKR2 before flashing firmware, I noticed a green LED light on the TFT35 board whenever the screen is attached to the SKR2.  After flashing that green light never comes on.

If someone has a working bin file, I would love to try it out, just to see if it might be something in my specific build.

Cr0nJ0b, 2021-12-16:
To add to this, I have since tried downloading the BTT Firmware from Github Marlin 2.0.9.2 bugfix.  compiled it stock with the only change being the uncomment of the #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER line.  It compiled properly and I used tried the default BIGTREE_SKR_2 build and the BIGTREE_SKR_2_F429 build.  neither firmware got rid of the "no printer attached" screen.  I also check continuity on the RS232 line from board to board and pin to pin.  The cable works and all pins are connected properly between the boards and match the pinout diagrams.

radek8, 2021-12-16:
The EXP1 and EXP2 cables are used for communication in Marlin.
The TFT connector is used for touch mode communication.
Send the used configuration files and photos of the current SKR and TFT connection.

radek8, 2021-12-16:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
is a driver for Marlin mode, this driver will work in simulated Marlin mode.
It has nothing to do with Touch Mode

radek8, 2021-12-16:
> the 5V pin always reads 1.3V for both SKR2 boards. Not sure if this is something that is just normal, but both boards had the same behavior.
>

What do you mean?

Cr0nJ0b, 2021-12-16:
I tested the +5V pin on the SKR2 board RS232 and it shows roughly +1.3V when plugged in.  The docs all say +5V.

Cr0nJ0b, 2021-12-17:
[Configuration_adv.h.txt](https://github.com/bigtreetech/SKR-2/files/7730961/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/bigtreetech/SKR-2/files/7730962/Configuration.h.txt)
[image]
[image]

Attached the Board pictures with a close up on the pins for the RS232.  and these are the "base" firmware that I just built from.  No real changes except for the TFT name.

radek8, 2021-12-17:
The red napaje dial jumper is in the wrong position

radek8, 2021-12-17:
Marlin mode is functional?

radek8, 2021-12-17:
Disconnect connectors EXP1 and EXP2. Does the TFT display light up and boot, only via the TFT-RS232 cable?

radek8, 2021-12-17:
You have to uncomment !!!
#define M115_GEOMETRY_REPORT
#define M114_DETAIL   
#define REPORT_FAN_CHANGE
#define HOST_ACTION_COMMANDS

You have to read the manual!
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Cr0nJ0b, 2021-12-17:
I walked through all of the options and thanks for the input.

1) I didn't understand the you were required to uncomment the lines listed, since the manual said that "in order the TFT firmware is able to provide ALL of it's functions/features, the following must be enabled."  I read this to mean that those items would not work correctly or at all rather than cause the printer to show as "no printer attached".  I made the changes as well as a number of the "optional" changes.  I assume I don't have to add them all.
2) I disconnected the EXP1 and EXP2 lines and the TFT will not light up.
3) Marlin mode is functional with the EXP lines attached.
4) the red jumper pins that you mention are for EITHER USB power OR 12/24V power for the motherboard.  For testing I was using USB power which is the correct location for the jumper.  I don't see anywhere in the documentation that the screen will show "no printer attached" with USB Power, but I have tried it with 12V power and i get the same error.

Cr0nJ0b, 2021-12-17:
I have it sorted out now.   I'm an idiot and just overlooked the TFT header.  I thought the SWD header was the TFT header for some reason and didn't check it again.  That's why it was showing low voltage.  I have some new issues with random noise on the TFT, but that's an issue for another day.  I'm glad to have the screen working and no issue with printer attached.  Thanks for the help.

closed 2021-12-17 by Cr0nJ0b

radek8, 2022-02-26:
These send configuration files

PaBu04, 2022-07-12:
> [Configuration_adv.h.txt](https://github.com/bigtreetech/SKR-2/files/7730961/Configuration_adv.h.txt) [Configuration.h.txt](https://github.com/bigtreetech/SKR-2/files/7730962/Configuration.h.txt) [...]

> [Configuration_adv.h.txt](https://github.com/bigtreetech/SKR-2/files/7730961/Configuration_adv.h.txt) [Configuration.h.txt](https://github.com/bigtreetech/SKR-2/files/7730962/Configuration.h.txt) [...]

You have to plug your display cable in the TFT slot, it's in the SWD slot.