bigtreetech/SKR-2 #93: SKR 2 w/ TMC2209 - TMC CONNECTION ERROR on 3 out of 5 Drivers

cjmalchow, 2021-12-17:
Hi,

I've run into a bit of a problem where my SKR 2 board is set for VCC power and is throwing low errors for me Y, Z, Z2(which isn't connected). X and E report back okay. I will post the full M122 below this message. Any help would be greatly appreciated! If you want to see the configuration.h, adv.h or any other of the files for compiling just let me know.

Thanks for anyone that helps!

Send: M122
Recv: 		X	Y	Z	Z2	E
Recv: Address		0	2	1	3	0
Recv: Enabled		false	false	false	false	false
Recv: Set current	1300	580	580	580	700
Recv: RMS current	1270	994	994	994	673
Recv: MAX current	1791	1402	1402	1402	949
Recv: Run current	22/31	17/31	17/31	17/31	21/31
Recv: Hold current	11/31	8/31	8/31	8/31	10/31
Recv: CS actual	11/31	0/31	0/31	0/31	10/31
Recv: PWM scale
Recv: vsense		0=.325	0=.325	0=.325	0=.325	1=.18
Recv: stealthChop	true	false	false	false	false
Recv: msteps		64	256	256	256	64
Recv: interp		true	false	false	false	true
Recv: tstep		max	0	0	0	max
Recv: PWM thresh.	316	1601	10621	10621	1075
Recv: [mm/s]		100	100	3	3	30
Recv: OT prewarn	false	false	false	false	false
Recv: triggered
Recv:  OTP		false	false	false	false	false
Recv: pwm scale sum	13	0	0	0	12
Recv: pwm scale auto	0	0	0	0	0
Recv: pwm offset auto	36	0	0	0	36
Recv: pwm grad auto	14	0	0	0	14
Recv: off time	0	0	0	0	0
Recv: blank time	24	16	16	16	24
Recv: hysteresis
Recv:  -end		2	-3	-3	-3	2
Recv:  -start		1	1	1	1	1
Recv: Stallguard thrs	0	0	0	0	0
Recv: uStep count	2	0	0	0	2
Recv: DRVSTATUS	X	Y	Z	Z2	E
Recv: sg_result	0	0	0	0	0
Recv: stst			*	*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:0B:00:00
Recv: 		Y	0x00:00:00:00	 Bad response!
Recv: 		Z	0x00:00:00:00	 Bad response!
Recv: 		Z2	0x00:00:00:00	 Bad response!
Recv: 		E	0x80:0A:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... Error: All LOW
Recv: Testing Z connection... Error: All LOW
Recv: Testing Z2 connection... Error: All LOW
Recv: Testing E connection... OK
Recv: ok

radek8, 2021-12-17:
Send the configuration. 
Send a photo of the pin configuration under the engine controllers. 
Send a photo of the underside of the engine controller

radek8, 2021-12-17:
If you swap X and Y, will X still work and Y will not work?

fab74560, 2021-12-25:
salut débranche la chauffe de la tète j'ai le même problème

J-charles-C, 2022-01-03:
> salut débranche la chauffe de la tète j'ai le même problème

Fab74560, peux tu préciser, Je galère sur ce problème depuis trois jours. Des drivers qui marchent très bien sur une autre machine me donne un TMC Error. Le sensorless homing ne fonctionne pas, le moteur part dans la mauvaise direction alors qu'avec un G1 cela fonctionne. J'en perd mon latin.

fab74560, 2022-01-04:
Bonjour j-charles 

Quand tu débranche le HE0 ca marche ces le HE0 qui a pt pour moi 

Si je le débranche tous marche bien mais la cartouche de chauffe 

Je te joint mon fichier de retour M122 

 

19:53:07.652 : No start signal detected - forcing start
19:53:07.661 : N1 M110*34
19:53:07.661 : N2 M115*36
19:53:07.661 : N3 M105*36
19:53:07.661 : N4 M114*35
19:53:07.661 : N5 M111 S6*98
19:53:07.661 : N6 T0*60
19:53:07.661 : N7 M20*22
19:53:07.663 : N8 M80*19
19:53:07.664 : FIRMWARE_NAME:Marlin bugfix-2.0.7.2.x (Jun 10 2021 14:06:38) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:SKR-2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
19:53:07.664 : N9 M220 S100*104
19:53:07.664 : N10 M221 S100*81
19:53:07.664 : Cap:SERIAL_XON_XOFF:0
19:53:07.664 : Cap:BINARY_FILE_TRANSFER:0
19:53:07.664 : Cap:EEPROM:1
19:53:07.664 : Cap:VOLUMETRIC:1
19:53:07.664 : Cap:AUTOREPORT_TEMP:1
19:53:07.665 : Cap:PROGRESS:0
19:53:07.665 : Cap:PRINT_JOB:1
19:53:07.665 : Cap:AUTOLEVEL:1
19:53:07.665 : Cap:RUNOUT:0
19:53:07.665 : Cap:Z_PROBE:1
19:53:07.665 : Cap:LEVELING_DATA:1
19:53:07.665 : Cap:BUILD_PERCENT:0
19:53:07.665 : Cap:SOFTWARE_POWER:0
19:53:07.665 : Cap:TOGGLE_LIGHTS:0
19:53:07.665 : Cap:CASE_LIGHT_BRIGHTNESS:0
19:53:07.665 : Cap:EMERGENCY_PARSER:1
19:53:07.665 : Cap:PROMPT_SUPPORT:1
19:53:07.665 : Cap:SDCARD:1
19:53:07.665 : Cap:REPEAT:0
19:53:07.665 : Cap:SD_WRITE:1
19:53:07.665 : Cap:AUTOREPORT_SD_STATUS:1
19:53:07.665 : Cap:LONG_FILENAME:1
19:53:07.665 : Cap:THERMAL_PROTECTION:1
19:53:07.665 : Cap:MOTION_MODES:0
19:53:07.665 : Cap:ARCS:1
19:53:07.665 : Cap:BABYSTEPPING:1
19:53:07.665 : Cap:CHAMBER_TEMPERATURE:0
19:53:07.666 : Cap:COOLER_TEMPERATURE:0
19:53:07.666 : Cap:MEATPACK:0
19:53:07.666 : area:{full:{min:{x:0.0000,y:0.0000,z:0.0000},max:{x:235.0000,y:235.0000,z:270.0000}},work:{min:{x:0.0000,y:0.0000,z:0.0000},max:{x:235.0000,y:235.0000,z:270.0000}}}
19:53:07.666 : N11 M111 S6*87
19:53:07.672 : X:0.0000 Y:0.0000 Z:0.0000 E:0.0000 Count X:0 Y:0 Z:0
19:53:07.672 : echo:DEBUG:INFO,ERRORS
19:53:07.672 : Begin file list
19:53:07.672 : N12 T0*9
19:53:07.672 : N13 M155 S1*82
19:53:07.679 : End file list
19:53:07.679 : echo:Unknown command: "M80"
19:53:07.679 : echo:DEBUG:INFO,ERRORS
19:53:16.684 : N14 M122*23
19:53:16.684 : X	Y	Z	E
19:53:16.684 : Address		0	0	0	0
19:53:16.716 : Enabled		false	false	false	false
19:53:16.716 : Set current	800	800	800	800
19:53:16.748 : RMS current	795	795	795	795
19:53:16.780 : MAX current	1121	1121	1121	1121
19:53:16.780 : Run current	25/31	25/31	25/31	25/31
19:53:16.780 : Hold current	12/31	12/31	12/31	12/31
19:53:16.813 : CS actual	12/31	12/31	12/31	12/31
19:53:16.813 : PWM scale
19:53:16.845 : vsense		1=.18	1=.18	1=.18	1=.18
19:53:16.877 : stealthChop	true	true	true	true
19:53:16.909 : msteps		16	16	16	16
19:53:16.941 : interp		true	true	true	true
19:53:16.973 : tstep		max	max	max	max
19:53:16.973 : PWM thresh.	82	82	98	274
19:53:17.005 : [mm/s]		120	120	20	30
19:53:17.037 : OT prewarn	false	false	false	false
19:53:17.069 : pwm scale sum	14	14	14	14
19:53:17.101 : pwm scale auto	0	0	0	0
19:53:17.133 : pwm offset auto	36	36	36	36
19:53:17.165 : pwm grad auto	14	14	14	14
19:53:17.197 : off time	4	4	4	4
19:53:17.229 : blank time	24	24	24	24
19:53:17.229 : hysteresis
19:53:17.261 : -end		2	2	2	2
19:53:17.293 : -start		1	1	1	1
19:53:17.293 : Stallguard thrs	0	0	0	0
19:53:17.325 : uStep count	40	40	40	40
19:53:17.325 : DRVSTATUS	X	Y	Z	E
19:53:17.357 : sg_result	0	0	0	0
19:53:17.389 : stst
19:53:17.421 : olb
19:53:17.453 : ola
19:53:17.485 : s2gb
19:53:17.517 : s2ga
19:53:17.549 : otpw
19:53:17.581 : ot
19:53:17.613 : 157C
19:53:17.645 : 150C
19:53:17.677 : 143C
19:53:17.709 : 120C
19:53:17.741 : s2vsa
19:53:17.773 : s2vsb
19:53:17.773 : Driver registers:
19:53:17.781 : X	0xC0:0C:00:00
19:53:17.790 : Y	0xC0:0C:00:00
19:53:17.797 : Z	0xC0:0C:00:00
19:53:17.805 : E	0xC0:0C:00:00
19:53:17.813 : Testing X connection... OK
19:53:17.821 : Testing Y connection... OK
19:53:17.829 : Testing Z connection... OK
19:53:17.837 : Testing E connection... OK

J-charles-C, 2022-01-04:
Salut,

cette carte est une misère sans nom.

Merci pour le tuyau, je vais tester.

cordialement

Eddiiie, 2022-02-16:
Any luck?

J-charles-C, 2022-02-18:
Bonjour,

J'ai trouvé une parade, j'ai passé deux steppers en 2208 et désactiver l'UART. 

J'attends un correctif pour mettre à jour la machine.

cordialement

fab74560, 2022-02-18:
Cool moi j’ai changer la carte pour une lerge  K-v2

Quand ils mon changer ma skr j’ai las moteur qui fonts un bruit  quand ils sont à l’arrêt    

Dit moi si sa marche avec le correctif 

merci