bigtreetech/SKR-2 #94: COMPILATION ERROR: 'MYSERIAL1' was not declared in this scope; did you mean 'MSERIAL'? && #error "SERIAL_PORT must be from 1 to 6. You can also use -1 if the board supports Native USB."

bfmvtony, 2021-12-18:
Hey guys!

I have recently downloaded last version of Marlin and tried to do some changes. To summarize, I modified an Artillery Sidewinder x1 with skr2 and the modifications I did were:

On plaformio.ini --> default_envs = BIGTREE_SKR_2

On config.h --> BOARD_BTT_SKR_V2_0_REV_B

I used the configuration example for X1, but I got these 2 errors

1. 'MYSERIAL1' was not declared in this scope; did you mean 'MSERIAL'?
2. #error "SERIAL_PORT must be from 1 to 6. You can also use -1 if the board supports Native USB."

It is quite strange because several days ago I did other things and it worked, but suddenly this appeared.

Could anyone help with this?

Thanks in advance!

bfmvtony, 2021-12-18:
It wa

bfmvtony, 2021-12-18:
it was an stupid thing, with this kind of board serial port must be 1, I read carefully and realized ;)

closed 2021-12-18 by bfmvtony

virus-sss, 2022-04-16:
я пытаюсь тоже самое сделать с платой MKS ROBIN PRO и получаю ту же самую ошибку