bigtreetech/SKR-2 #99: SKR V2 doubles movement after print start

boxburn, 2021-12-25:
I made a post in the main branch of [Marlin](https://github.com/MarlinFirmware/Marlin/issues/22996) & another user [here](https://github.com/bigtreetech/SKR-2/issues/96) has a similar issue.

What I have noticed is that the msteps go from 16 down to 8 along side a few other things changing. This doesn't always happen as soon as the print starts, one attempt got half way through the first layer before doubling movement, another it did it as soon as it had homed.
[image]
[image]

As requested by a reply in dydyamotya's post, here's the M122 before the print and after the issue and pictures of my TMC2209 and board.

<details>
  <summary>M122 Before</summary>
  
>M122
SENDING:M122
		X	Y	Z	E
Address		0	0	0	0
Enabled		false	false	false	false
Set current	800	800	800	800
RMS current	795	795	795	795
MAX current	1121	1121	1121	1121
Run current	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31
CS actual	12/31	12/31	12/31	12/31
PWM scale
vsense		1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true
msteps		16	16	16	16
interp		true	true	true	true
tstep		max	max	max	max
PWM thresh.
[mm/s]
OT prewarn	false	false	false	false
triggered
 OTP		false	false	false	false
pwm scale sum	14	14	14	14
pwm scale auto	0	0	0	0
pwm offset auto	36	36	36	36
pwm grad auto	14	14	14	14
off time	3	3	3	3
blank time	24	24	24	24
hysteresis
 -end		-1	-1	-1	-1
 -start		1	1	1	1
Stallguard thrs	75	75	0	0
uStep count	8	8	8	8
DRVSTATUS	X	Y	Z	E
sg_result	0	0	0	0
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:0C:00:00
		Y	0xC0:0C:00:00
		Z	0xC0:0C:00:00
		E	0xC0:0C:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
</details>

<details>
  <summary>M122 After</summary>
  
>M122
SENDING:M122
		X	Y	Z	E
Address		0	0	0	0
Enabled		true	true	true	true
Set current	800	800	800	800
RMS current	1436	1436	1436	1436
MAX current	2025	2025	2025	2025
Run current	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31
CS actual	31/31	31/31	30/31	31/31
PWM scale
vsense		0=.325	0=.325	0=.325	0=.325
stealthChop	true	true	true	true
msteps		8	8	8	8
interp		true	true	true	true
tstep		max	max	max	max
PWM thresh.
[mm/s]
OT prewarn	false	false	false	false
triggered
 OTP		false	false	false	false
pwm scale sum	80	56	35	67
pwm scale auto	0	0	0	0
pwm offset auto	79	58	31	67
pwm grad auto	28	23	14	26
off time	3	3	3	3
blank time	36	36	36	36
hysteresis
 -end		-3	-3	-3	-3
 -start		6	6	6	6
Stallguard thrs	75	75	0	0
uStep count	16	144	16	80
DRVSTATUS	X	Y	Z	E
sg_result	32	148	0	18
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:1E:00:00
		Y	0xC0:1F:00:00
		Z	0xC0:1F:00:00
		E	0xC0:1F:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
</details>

boxburn, 2021-12-26:
I had a print do the same again so I paused the print, sent M500, homed all axis and was then able to print the full Benchy test with no issues.

radek8, 2021-12-28:
It is strange that when the error occurs, the registry address changes.
Can you send configuration files?

boxburn, 2021-12-28:
If you need it sending as a file, I'm not sure how to do that and just going out to work now. Thanks
<details>
  <summary>Config.h</summary>

>
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 */
#define CONFIGURATION_H_VERSION 02000903

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some useful links to help get your machine configured and calibrated:
 *
 * Example Configs:     https://github.com/MarlinFirmware/Configurations/branches/all
 *
 * Průša Calculator:    https://blog.prusaprinters.org/calculator_3416/
 *
 * Calibration Guides:  https://reprap.org/wiki/Calibration
 *                      https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 *                      https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 *                      https://youtu.be/wAL9d7FgInk
 *
 * Calibration Objects: https://www.thingiverse.com/thing:5573
 *                      https://www.thingiverse.com/thing:1278865
 */

//===========================================================================
//========================== DELTA / SCARA / TPARA ==========================
//===========================================================================
//
// Download configurations from the link above and customize for your machine.
// Examples are located in config/examples/delta, .../SCARA, and .../TPARA.
[... 2883 lines cut ...]

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Set to 0 to turn off servo support.
 */
//#define NUM_SERVOS 3 // Note: Servo index starts with 0 for M280-M282 commands

// (ms) Delay before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Edit servo angles with M281 and save to EEPROM with M500
//#define EDITABLE_SERVO_ANGLES

// Disable servo with M282 to reduce power consumption, noise, and heat when not in use
//#define SERVO_DETACH_GCODE

</details>