bigtreetech/SKR-2 #101: SKR2 STM32F429VGT6 does not link SD card as drive on pc

mikeww1, 2022-01-02:
I have switched from a skr 1.3 to 2 form y anet a8plus and TMC 2209 drivers.
I used marlin 2.0.9.3 because in the release notes it says: New Board : BigTree SKR 2 with F429 (#23177)
So I guess this is the right release for my build.
I used the config files from this site to adapt my config files.
Now I have some questions:

What is the difference between the USB build and the non USB build ?

I tried both. With the USB build I could move the axis and talk to the printer and the SD card was mapped as a drive on my PC.
But I could not access the file system.

With the non USB build I could not even move the axes from the Marlin display. I could connect to the PC but could not control the printer. So I guess USB is the right build for me. What do I have to do to be able to use the SD card as a drive like with the SKR 1.3

With both build I was experiencing reboots every few minutes. I found this thread 
https://github.com/bigtreetech/SKR-2/issues/82
that deals with that problem. The recommendation is to disable:

Configuration_adv.h
#define USB_FLASH_DRIVE_SUPPORT
#define USE_OTG_USB_HOST
#define NO_SD_DETECT
#define MULTI_VOLUME

Is this the recommended action ? For what use is the other USB connector ?

Any help is very welcome !

mikeww1, 2022-01-03:
Ok I am one step further. When I use the USB build and by commenting out the above options I get a connection with the SD card and can see the files. Nevertheless this does not work reliably. When I use the reset button on the display the drive stays disconnected. Only when I pull out and in the USB connector on the PC the drive gets reconnected. This worked better before.
The remaining problem is that I cannot control the steppers. Not via Marlin Display nor via PC. I tried the other build without USB with the same result. I guess I have to start from scratch....

mikeww1, 2022-01-04:
OK I could solve the last problem. Although there is a new option in Marlin that should reset the EEprom after flashing, that feature does not seem to work. After I manually reset the EEprom I could move my axes as exspected. So I am set except for the non working extra USB connector.

closed 2022-01-04 by mikeww1

ANT-NONIUS, 2022-02-06:
Got the same problem.
Changing 'default_envs = BIGTREE_SKR_2_F429_USB' in platformio solved it to me.
Some times you have to reconect de USB cable for the SD card to show up in windows.