bigtreetech/SKR-2 #105: Bootloader for SKR-2 STM32F407

norellmans, 2022-01-16:
Hi, i manged to destroy the STM32F407 while fiddeling on the potentiometer for a driver and have now swapped it to a new one.
I cant really find a reliable source for the bootloader. I have tried some for the PRO model which seems to work but are ending up with connection issues after a succesfull flashing of marlin.

Could someone either extract their bootloader with a st-link or could the autors just add the original bootloader to the repo?

thanks in advance

Danangchandra, 2022-02-11:
please send email to service004@biqu3d.com, and they will send you bootloader file and step by step guide to flash it into your board.
actually yesterday I was burn 32f407 chip then replace it and now my board working well

waynemurray74, 2022-06-07:
I started looking for this today, found this post, emailed them, and just got the bootloader!!! Thanks so much for sharing this!