bigtreetech/SKR-2 #107: CoreXY - Y autohome bad direction

vinoudropdrop, 2022-01-20:
Hi!
I am using a CoreXY printer.
The configuration is : SKR2.0 Rev A unmodified + TMC2209 (grounded wires) + BL-Touch.
I think it is important to explain everything that happened to understand what is going on.
I wanted to install UBL. But I am not used to it, so I preferred to proceed steps by steps for the configuration. So I set up everything in bilinear the time to do the tests and to configure everything well. Because I have ever used bilinear.
Until then, everything was working. 
The X and Y movements were coherent, autohome worked, the palpation of the grid was ok, the babystepping too and even a successful first impression. 
So I recompiled marlin by switching from bilinear to UBL.
I started the configuration by doing M502, M500 then M501.
When it came time to homing in G28, the X displacement goes to the origin but the Y displacement goes in the opposite direction, that is to say that the nozzle moves in positive in the X axis and in the Y axis. And since this manipulations, I can't do a simple homing no more...
I tried many things, even go back to an old version compiled in bilinear, nothing works, the Y homing always goes in the wrong direction. 
So I ask for help. 
Thanks a lot.
PS : I can attach a YouTube video of the homing if needed.

radek8, 2022-01-21:
// Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
#define INVERT_X_DIR true
#define INVERT_Y_DIR false
#define INVERT_Z_DIR false

vinoudropdrop, 2022-01-21:
Thank's for reply, but issue is solved. It was not dir. Dir was working well. It was because of a endstop wire not working. I've forgot to close the issue.

closed 2022-01-21 by vinoudropdrop