bigtreetech/SKR-2 #108: Unable to find updated files that fix missing pin definitions

DaBoss1325, 2022-01-21:
I am unable to compile Marlin for my SKR 2.0 due to undefined pins, as mentioned in issue #23221. If I understand what was said correctly, the issue was fixed, but I am unable to find the files that I need to download to replace the improperly working files. I am completely confused by how Github seems to work, so it could be something super obvious that I'm missing. I had to go to Google just to figure out how to download anything from here, let alone find the files that I need. I'm so confused. Please help.

vinoudropdrop, 2022-01-21:
You have : 
`BIGTREE_SKR_PRO  FAILED`
But you should have :
``BIGTREE_SKR_2  FAILED``
Are you sure you are compiling 
withis this command : 
#define MOTHERBOARD BOARD_BTT_SKR_V2_0_REV_A
or
#define MOTHERBOARD BOARD_BTT_SKR_V2_0_REV_B
?

DaBoss1325, 2022-01-21:
I do not have that error. I am saying that I have the same "TOUCH_MISO_PIN not defined" error that the OP in #23221 was having, except on the SKR 2.0. I am assuming that the fix one of the commenters mentioned would also work for my board, but I can't find the updated files!

DaBoss1325, 2022-01-21:
Here's the error I'm getting: https://pastebin.com/CM66dtRC