bigtreetech/SKR-2 #109: Random Reboot

Daniele981, 2022-01-23:
Hi everyone,
I'm building a Mk3 clone with a skr 2 and tft 50 display.
My 3d printer reboot randomly after about 10 m and obviously I'm not able to print nothing...
I checked the voltage of Prusa Stock Psu and it's ok (24v).
I think that the problem is 5v voltage of the motherboard...
Has anyone had this problem before?

Daniele981, 2022-01-26:
I tried with my old skr 1.4 and the printer doesn't reboot!
So the PSU is not the problem...
I think there is a problem with the motherboard and also with marlin! I don't think I will mount skr 2 again until new updates come out!

tankslappa, 2022-02-02:
I have the same issue with the STM32F429 based SKR2.
I discovered that if you remove the firmware microSD card the reboot issue stops.

derbenner, 2022-02-02:
proved! without micro sd they dont reboot.

Eddiiie, 2022-02-16:
Same here.  lol  WTF

codephotog, 2022-02-16:
I can confirm that if I leave the microSD card slot in the printer it will restart constantly restart. I didn't time it, but 180 seconds seems about right. Just long enough to get the hotend up to temp and the bed most of the way there. Once microSD is removed directly from the board it will stay booted. 
The board is connect to a stock Ender 5 Plus with no changes to PSU or bed mosfet. BTT 3.5 touchscreen with SD card in slot.

Eddiiie, 2022-02-16:
So basicly this board is useless (must use octoprint).
I cannot get the stepper motors to move at all.
There was the Rev A board problem too.

Run away!!

tankslappa, 2022-02-16:
> So basicly this board is useless (must use octoprint). I cannot get the stepper motors to move at all. There was the Rev A board problem too.
> 
> Run away!!

Yes there was an issue on early boards, but that has been fixed. They offered free exchange.

I've printed with mine via octoprint and direct usb to my laptop, so I have no idea why you think that's an issue.

I'm using 2209 and 8845 drivers, no issue with either. Have you tested those motors and leads on anything else before?

Eddiiie, 2022-02-16:
I opened new issue on the steppers not moving.

https://github.com/bigtreetech/SKR-2/issues/122

Thank for the help!

binhamed84, 2022-02-16:
I got random rebooting every 1 min on my BTT SKR 2 rev B F429, but I fixed that by editing the files and make the correct configuration and after updating the BTT TFT35 V3.0 E3 touch screen, everything looks amazing and works great.

Eddiiie, 2022-02-16:
> but I fixed that by editing the files and make the correct configuration 

What do you mean by this?

yzheka, 2022-05-12:
I confirm the same issue. if FIRMWARE.CUR presents on micro sd card then it randomly reboots.

yzheka, 2022-09-06:
> I had the same issue, but in my case i turned off the minigames and all the animations - successfully.

I don't think this depends on marlin firmware features. I use klipper firmware and have rebooting when sd card inserted

lcortese, 2024-03-31:
I had a similar issue with a SKR Mini E3 v2. The printer is restarting at random times. I tried to reproduce the issue: at some point during the printing, if I removed the SD card, the exact same symptom occurred.

So I guess the problem could be related to a random disconnection of the SD module. Take this as a theory, because I am not an electronics expert.