bigtreetech/SKR-2 #111: BIGTREETECH SKR 2 board with a STM32 F429 Marlin  uncontrolled RESET !!!

qulet, 2022-01-24:
I have problems with my BIGTREETECH SKR 2 board with a STM32 F429 chip on it.
I am using the current "Marlin-bugfix-2.0.9.2.x" branch from just now.
Performs builds for the BIGTREE_SKR_2_F429 environment.
Unfortunately, Marlin with a connected printer on TCM2208 seriall controllers, after executing "Home", there is an uncontrolled RESET
This is my second board advertised with the F429 processor. ...... The complaint is an uncontrolled resetting of Marlin.

I don't know where the problem is ...
hardware or software
[Config.zip](https://github.com/bigtreetech/SKR-2/files/7929434/Config.zip)
?

Dennis02121978, 2022-01-26:
I have the same Problem.

qulet, 2022-01-27:
Is there anyone who tested BTT SKR2 ReV B F429 with TMC UART drivers and Marlin-bugfix 2.09.2x compilation for F429?
  If Yes, please share your comments.

mslucass, 2022-01-27:
No but BTT SKR2 Rev B F429 with TMC2209 I don't have reboots

qulet, 2022-01-27:
Thank you for the info on TMC2209
I have Bigtreetech TMC2208 Uart Marlin drivers, it does not show a communication problem! And yet it resets. Can someone tell you what to change in the pins_BTT_SKR_V2_0_common.h file so that there is no restart?

qulet, 2022-01-27:
mslucass
Can you share your configuration files?
Configuration_adv.h
Configuration.h
pins_BTT_SKR_V2_0_common.h

mslucass, 2022-01-27:
[pins_BTT_SKR_V2_0_common.zip](https://github.com/bigtreetech/SKR-2/files/7950304/pins_BTT_SKR_V2_0_common.zip)
[Configuration + adv.zip](https://github.com/bigtreetech/SKR-2/files/7950311/Configuration.%2B.adv.zip)

One small problem is that I don't have a backup of my config files before I changed them yesterday.
Yesterday I have added support for a bltouch and smart filament sensor. Both changes are untested.
The changed lines against a default are with //maurice on the end.

qulet, 2022-01-27:
mslucass
Thanks !

qulet, 2022-01-27:
mslucass
I have modified your files for my printer and it is a success !!!!! Merlin doesn't reset.
THANK YOU VERY MUCH

qulet, 2022-01-28:
One resets, but after a long time which prevents printing anyway: ((((

qulet, 2022-01-28:
Found the reason for the Marlin reset.
Even though there is no compile error, "Zservo probe" "Laser control" cannot be run at the same time, this is the reason for the reset.
Compiling "Fix-Mounted Probe" and "Laser control" does not restart Marlin

qulet, 2022-01-28:
Running the #define LASER FEATURE function causes the reset.

tankslappa, 2022-01-29:
I'm having the same approx 180 seconds reboot.
BTT SKR2 Rev B F429 with 2 x TMC2209 and 2 x 8845
I'm not using LASER_FEATURE or auto bed levelling.

M111 S7 doesn't give any clues, there is no error message from Marlin.

If I remove the firmware microSD from the SKR2 (it's only needed when the SKR copies the firmware.bin to its internal flash memory), it stops rebooting. So it gives me a work around, but there is obviously some hardware clash.

Dennis02121978, 2022-01-29:
you must remove the mini-sd onboard then no reboot

tankslappa, 2022-02-02:
> you must remove the mini-sd onboard then no reboot

Yes, this is the workaround I mentioned. It's still a bug though.

qulet, 2022-02-02:
I confirm, I tested only on the plugged in TMC and the LCD display with an active laser function for 30 minutes there was no reset :)))

Randwic, 2022-04-14:
I have the same problem, when I putt an sd card on motherboard, I have a restart every 180 sec.

Skr2 f429
Tmc 2226
Marlin 2.9.3

waynemurray74, 2022-06-08:
You can change to the correct chipset in marlin 2.1. Here is a video showing how. Should help fix your issues....
https://youtu.be/czm5D3-DPyk